ArduPilot/ardupilot #2091: SD Card fails to mount after the soft reboot, after reflashing

jmachuca77, 2015-04-12:
With master, after reflashing the firmware with APM planner, when the pixhawk soft reboots it fails to mount the SD card with the following output on the console:
```
�[boot] Initialized SPI port 1 (SENSORS)
[boot] Initialized SPI port 2 (RAMTRON FRAM)
[boot] Initialized SPI port 4
[boot] Initialized SDIO
[rgbled] on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
[init] looking for microSD... 
nsh: mount: mount failed: Out of memory
Trying format of microSD 
nsh: mkfatfs: mkfatfs failed: Operation not permitted
format failed 
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[init] USB interface connected 
Opening USB nsh 
nshterm [11:70]

NuttShell (NSH)
nsh> 
```

If I do a hard reboot (cycle power) then it works correctly.

OXINARF, 2017-06-02:
@jmachuca77 This is quite old and probably hard to debug after so much time. Can it be closed?

EShamaev, 2017-06-28:
I had the same issue and the reason was bad SD card.

closed 2018-01-11 by lthall