ArduPilot/ardupilot #2093: Onboard Linux companion computer eventually gets locked out of Pixhawk USB interface

akwatson, 2015-04-12:
Using the latest APM Rover firmware (ver. 2.49) on a Pixhawk connected to a Raspberry Pi 2 (RPi2) via the Pixhawk's USB to the RPi2's USB, I can lock myself out of the Pixhawk's serial port despite using standard python calls to open a serial connection (using mavutil's mavlink_connection()) and proper calls to close() the serial connection in response to a SIGINT signal during the python script shutdown.  Everything works fine for awhile while sending/receiving mavlink telemetry data using this method, but enough opens and closes to the Pixhawk (/dev/ttyACM0) and eventually mavutil's recv_msg() will only return "None" and I'm forced to push a paperclip into the side of the Pixhawk to do a hardware reset.  I think this is probably a bug, but would be happy to hear otherwise.  I put together code to reset the RPi2's USB port occupied by the Pixhawk (simulating an unplug-replug), but it does not cause any change in what mavutils' recv_msg() returns (it remains "None").  I think this is an issue because it raises the possibility that the onboard computer won't be able to get back into the Pixhawk after a python script relaunches and tries to issue an RTL to the Pixhawk while airborne.  Thoughts?

silviorevelli, 2015-05-09:
I'm using oDroid with this architecture (see www.4gmetry.com). Your problem is composed by two sub-problems: 1) sudden disconnection 2) difficult to reconnect. I never experience the first point, but I can 100% confirm the second. Once the connection is lost (maybe when I shut down the companion computer, or physically unplug the USB) it's pratically impossible to re-establish the connection without rebooting Pixhawk.

akwatson, 2015-05-15:
Interesting and, just to clarify, my experience is exactly the same as your own.  The RPi2 has never gotten a "sudden disconnection" while actively sending/receiving data to the Pixhawk.  Every time the RPi2 been locked out of the Pixhawk, it has been when trying to reconnect to the Pixhawk -- and only after multiple open/closes of the Pixhawk using the serial USB interface.  If this isn't a well known issue (and I'm not doing something wrong), I could put together some test code to try and deliberately put the Pixhawk into this locked up state if it'd help diagnose what's happening. 
  
Here was how I resolved the issue:   I crimped a cable linking the Pixhawk's 6-pin Telemetry 2 port to the Raspberry Pi 2's GPIO pins (dev/ttyAMA0) and stopped using the USB cable for telemetry.  I made this change over a week ago and, with the exact same python code, have yet to see the RPi2 get locked out of the Pixhawk.  When I was using the USB cable, I was having to hard-reset the Pixhawk about a dozen times a day.

akwatson, 2015-05-15:
Here's a short Python script that, at least on my OSX machine, will pretty quickly put my USB-connected Pixhawk in a locked out state. Welcome any comments. 

pastebin.com/Fdi2qurk

Naterater, 2019-01-07:
@akwatson, do you have any updates on this?

peterbarker, 2021-07-30:
No updates after requested a 30 months ago.... closing.

Everything's been rewritten....

closed 2021-07-30 by peterbarker