ArduPilot/ardupilot #2094: Pixhawk and EMAX ESC

All4U, 2015-04-12:
I wonder the need of raising an issue...about Emax ESC and the Pixhawk:
- When I plug the ESC direct to X8R receiver, power up, hearing some tone from motor, raise up the throttle and the motor spin: OK
- When I plug the ESC through mainout of Pixhawk, after Arming the motor, the ESC seem no wake up and the motor didnt spin! (I can see the output value via APM planner when throttle is raise up)
- I did some gooogle and see that many of us faced with this problem, they guess the causes may lied in hardware like Voltage level (pixhawk output) or in the Firmware like the required pulse to wakeup the Emax ESC before it timeout, etc....but actually, no such good solution...
So, please here any one had come over this can share the solution?

Best regards,

R-Lefebvre, 2015-04-12:
This is not an APM software issue, nor is it an issue with the Pixhawk hardware.  This is an issue with the Emax ESC.

rmackay9, 2015-07-22:
I've reopened this issue at least as a place holder.  The issue appears to be that the Emax ESC gets confused by the high signal that the Pixhawk outputs at startup.  The APM2 for example outputs a low signal on startup.

I'm afraid it's a fairly low level change in the PX4Firmware layer which I'm not personally going to be able to resolve I think.  Perhaps another developer can look into it or maybe the EMAX ESC people can resolve the issue on their side.

There's a related issue to the Parachute here.
https://github.com/diydrones/ardupilot/issues/1239

alduxvm, 2015-07-28:
To solve this issue, remove the 2 resistors marked in the picture and that will be it, Randy and Robert were right, this is a ESC hardware issue... 

[image]

magicrub, 2015-07-28:
The two components in your circle look like a resistor and a capacitor, not
two resistors.

alduxvm, 2015-07-28:
Yes, remove those two components...

magicrub, 2015-07-28:
Solved, and also a duplicate of https://github.com/diydrones/ardupilot/issues/2033

closed 2015-07-28 by magicrub

estechnical, 2015-12-02:
Hi, I thought I should follow up on this issue as it seems people are modifying their EMax ESCs...

I have modified mine and they run fine now - I removed the resistor and capacitor after finding the signal level was very low and getting the meter out.

After making this modification however I would now suggest an extra step/improvement to the solution as this modification makes the ESCs no longer work with the EMax program card. To make the program card work again, it is necessary to add a 10k pull down resistor to the signal so that the bidirectional serial communication can work with the program card.

Also, just an observation but possibly important, without the pull down resistor the signal line is left electrically floating if the signal lead is unplugged and is only connected to the pin of the MCU in the ESC.

The best solution to this is:
replace the bottom resistor (circled in the photo above) with a 10k resistor
*and*
replace the resistor immediately above the circled components with a low value eg 100R or 220R.
The capacitor should also be removed as per the suggestion above.

alireza-manzoori, 2016-08-05:
Interesting update on the EMAX ESC issue with Pixhawk: **The issue is apparently resolved in the new generation of EMAX Simon ESCs!**

I was making a new quadrotor and again decided to use Emax Simon ESCs despite their compatibility issue with Pixhawk (as they are by far the cheapest and most available ESC in the local stores here). Before ripping the heat shrink on the ESCs to remove the resistor and capacitor as I had done before, I decided to test them. So I attached them to Pixhawk and the motors started spinning right away!

I think EMAX has modified the ESCs to resolve this problem finally, but maybe it's still too soon to jump to this conclusion. I have tested Emax Simon 20A ESCs with Pixhawk, Copter Firmware V3.2 & 3.3 both worked fine.

Steadyspike, 2017-10-01:
I have same problem with Emax Lightning 30A BLHeli and Pixhawk. Does someone know which components should I replace? Find attached a photo.

[image]
[image]

9friday, 2020-09-15:
I want to use the Emax BLHeli Series 30A ESC(https://emaxmodel.com/catalog/product/view/id/101/s/blheli-series-30a-esc/) with the Pixhawk. 
Is the issue with Pixhawk resolved for all EMAX ESCs or only specific models?
This link(https://ardupilot.org/copter/docs/connect-escs-and-motors.html) still mentions that Pixhawk has an issue with EMAX ESCs.

Any help would be appreciated.
Regards.

rmackay9, 2020-09-15:
@9friday, It seems that the underlying issue is with the ESC itself.  I don't think we have heard if the manufacturer has fixed the issue.  If you do buy them please tell us if they seem to work or not.

mbuzdalov, 2021-01-24:
Just for an information, I can confirm that ESCs called "Emax Formula 45A BLHeli32 32Bit" work quite well with Pixhawk.

mateoc6, 2022-03-30:
> Just for information, I can confirm that ESCs called "Emax Formula 45A BLHeli32 32Bit" work quite well with Pixhawk.

Hi Mbuzdalov,
I recently purchased the exact same ESCs for my drone and couldn't seem to get my motors running through them. I was wondering what you did when connecting it to the cube/pixhawk. I currently have the white wire/signal wire connected to signal on the main out signal pin. Then the same thing for the ground wire.

rmackay9, 2022-03-30:
@mateoc6, maybe the safety switch is on?  Try setting BRD_SAFETYENABLE = 0.  Just a random guess..

mateoc6, 2022-03-30:
> @mateoc6, maybe the safety switch is on? Try setting BRD_SAFETYENABLE = 0. Just a random guess..

I will check that option, that's a good possibility. Thank you

mateoc6, 2022-03-31:
> @mateoc6, maybe the safety switch is on? Try setting BRD_SAFETYENABLE = 0. Just a random guess..

I took a look at the setting and it was already set to 0. I will probably search the ardupilot page to see if anyone has posted this before, if not start a new post about it.

mituritsyn, 2022-08-22:
> 

I've faced similar issue with another ESCs. This is due to low resistance on esc signal pin lead to PWM amplitude lowering so ESC couldn't recognise it