ArduPilot/ardupilot #2095: Copter: support triggering two separate cameras

rmackay9, 2015-04-12:
This would require:
1. a "frontend/backend" split of the camera library.
2. a new aux switch option for "Camera Trigger 2"
3. modify the do-digicam-control to accept a camera instance (at this moment param7 could be used)

markwbrown, 2016-07-02:
Any progress here?

amilcarlucas, 2023-09-13:
I think this is now implemented. Can you test ArduCopter 4.4.1 or later?

rmackay9, 2023-09-13:
This is not actually complete yet but should be soon.

Part1 is done.
Part2 will probably never be done because we will instead rely on GCS buttons.
Part3 is partially done.  The mavlink change is in (https://github.com/ArduPilot/mavlink/pull/328) but we still need to do the ardupilot flight code change

rmackay9, 2023-09-22:
Part 3 is done in master now so I will close this.

closed 2023-09-22 by rmackay9