ArduPilot/ardupilot #2096: SITL Vagrant instructions broken - need to confirm new instructions

hamishwillee, 2015-04-12:
The commit [Support building of Pixhawk firmware under Vagrant/VirtualBox](https://github.com/diydrones/ardupilot/commit/33af275220a1dcfb553ed67b8f54a273acdc85e6#diff-23b6f443c01ea2efcb4f36eedfea9089) changed the requirements for using Vagrant so the instructions in [/vagran/README.md](https://github.com/diydrones/ardupilot/blob/master/Tools/vagrant/README.md) and [Setting up SITL using Vagrant](http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/setting-up-sitl-using-vagrant/) (wiki) no longer work.

Specifically the vagrantfile now requires that you have the repos for PX4Firmware, PX4NuttX and uavcan downloaded alongside ardupilot.

Once these git rhave been pulled, ``vagrant up`` gets a bit further.

Then it fails with:
```
"rsync" could not be found on your PATH. Make sure that rsync
```

If rsync is required, is there a particular source we should get this from? e.g. http://www.rsync.net/resources/howto/windows_rsync.html ?

Are all these extra dependencies really necessary? If not, could they somehow be made conditional/optional?

Are there any other dependencies?

mrpollo, 2015-04-13:
re rsync: I found this on Vagrants site regarding rsync synced folders http://docs.vagrantup.com/v2/synced-folders/rsync.html

hamishwillee, 2015-04-14:
The point of having a SITL vagrant file is to make setting things up reproducible and *easy*. Adding rsync dependency as well as extra github dependencies makes that impossible. 

I've  got this working but it doesn't come up "cleanly" - see #2109. I have also created #2110 to remove most of the README.md instructions, replacing them with a link to the ArduPilot HowTo.

I've updated the howto and requested review from the wiki editors.

andyp1per, 2015-05-08:
I made the original change, I'm happy to take a look and see if it can be made more resilient to different setups.

hamishwillee, 2015-05-08:
@andyp1per . Just to be clear, your changes work, but they made the instructions more complicated on windows because of the additional rsync dependency. 

I don't think this is needed - you could have the file you need from git cloned under the vagrantfile directory - this way they are automatically shared with the host (and the vm at /vagrant). If this is a problem from the vagrantfile location, then one alternative is to clone them in the VM.

Thanks very much for looking into this.

hamishwillee, 2015-05-08:
And the instructions are fixed, so I will close this. Andy, if you change things to do it another way, can you please either update the instructions, or add my name to the change request so I can do so.

closed 2015-05-08 by hamishwillee

andyp1per, 2015-05-09:
The reason for rsync is because NuttX requires symlinks and these are pretty much broken on virtualbox on windows. I spent a long time looking into various methods of making this work, but in the end rsync was the only reliable one. The only other option is to actually git clone inside the VM, but that slightly defeats the point of using vagrant where you want to share the FS with the host. But if folk feel that is a more robust answer I could change it to work that way.

hamishwillee, 2015-05-11:
Hi Andy, 

I had hoped you hadn't done the "due-diligence" here. Given that you've investigated and found that this is the best way lets leave "as is".

Cheers
H

PS The guess part of the problem here is that the Vagrantfile doesn't have a clear stated purpose and owner - so anyone can change it to suit their purposes and this might not "do the right thing" by other cases. In any case, I will further update the instructions on using SITL so they provide an easier option for developers using ardupilot rather than working on ardupilot (https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/88)

andyp1per, 2015-05-11:
I'm beginning to think that cloning PX4Firmware, PX4NuttX and uavcan inside the VM might be the right answer. You pretty much don't care unless you are actually doing development on these, which almost no-one is. It would then mean that vagrant is easier to get up and running and avoid the symlink issues. Even with my change I see permissions issues on make px4-clean in PX4Firmware.

hamishwillee, 2015-05-11:
Sounds great to me. 
For developers, what else extra would they need? If you clone inside the /vagrant directory from inside the VM the git tree is still available in both windows and Linux and presumably isn't broken unless someone explicitly plays with a symlink in windows?

andyp1per, 2015-05-12:
The cloned repos would only be available inside the VM. It's the sharing that causes the issues.

hamishwillee, 2015-05-12:
Please do "whatever is best for developers OF ArduPilot and SITL". As long as the behaviour for developers is documented in [Setting up SITL using Vagrant](http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/setting-up-sitl-using-vagrant/) then it will be fine for me.

That might seem a little bit of a backtrack. Reason is we now have another Vagrant environment for *users* of SITL who just need a simulated vehicle. This is a pre-built zero-configuration image and doesn't share anything with the host - so it is fast and easy to use. We still need something for developers, and that is this image that you've been working on.