ArduPilot/ardupilot #2097: Copter: twitches in flight if using Ublox M8

rmackay9, 2015-04-13:


ThorstenDP, 2015-04-13:
Randy, thanks!
For details, reference as well as some discussion see:
http://ardupilot.com/forum/viewtopic.php?f=80&t=10155
https://groups.google.com/d/msg/drones-discuss/r7gREVd3Ilw/MlERC5JHlwcJ
There are reports of M8N twitches from others as well at DIYDrones.
Would be great if the cause could be identified (not to mention a fix :-) )

robustini, 2015-04-13:
Where is this "M8N twitches"?

https://www.youtube.com/watch?v=HnlBOVPkEqI

https://www.youtube.com/watch?v=i5q1iM6_37U&feature=player_detailpage#t=75

Marco

ThorstenDP, 2015-04-13:
Hi Marco,

well, there are no visible twitches in your videos.
The thing is they only occur from time to time. And the cause seems to be missing GPS samples. You can see it in the log I posted at http://ardupilot.com/forum/viewtopic.php?f=80&t=10155
I am pretty sure I configured the GPS correctly. I tried several versions (VirtualRobotix, CSG Shop, Drotek) on several copters with original PixHawks as well as the DroPix. Hence, I am pretty sure it is not some other problem.
Anyway, it would be great if you could post you configuration settings (and if someone could place them in the Wiki).

Thanks and best regards,
Thorsten

robustini, 2015-04-13:
Reset "M8N" to the "U-Blox" default, save the parameters then repower the unit.
I've setup my M8N with the following configuration parameters:
PRT (ports): UART 1 -> Protocol out- UBX
Baudrate: 38400
RATE (Rates): Measurement Period 200 ms (5 Hz)
NAV5 (Navigation 5): Dynamic Model - Airborn 4G
Fix Mode: 3 - Auto 2D/3D

Everything else at default.

ThorstenDP, 2015-04-13:
Thanks! Yes that's what I am using as well.
What values are you using for EKF_POS_DELAY and EKF_VEL_DELAY? Paul Riseborough suggests to set them to 120: https://groups.google.com/d/msg/drones-discuss/r7gREVd3Ilw/08otn4tbNZgJ
Interestingly, Paul reports that "The M8 had slightly less velocity noise." compared to the 6H.

SuperSharpShooter, 2015-04-27:
I've just installed an m8n from rtfquads, and now I'm getting the twitch in all GPS modes. Previously I had a neo-6m on here and no twitch.

SuperSharpShooter, 2015-05-10:
Turns out my GPS wasn't writing the config changes to ROM properly. Have re-configured it to have Marco's settings and made sure they saved and the twitch looks to have gone.

rmackay9, 2015-08-04:
@SuperSharpShooter, All,
We think that these issues may be resolved by using the later copter beta (AC3.3-rc8) so if anyone is still seeing these problems and is comfortable flying a beta version of the software (which has been tested quiet thoroughly but is still more risky than using AC3.2.1), then could you you try the beta to see if it helps?
http://diydrones.com/forum/topics/copter-3-3-beta-testing

WickedShell, 2015-08-04:
Particularly with the use of http://copter.ardupilot.com/wiki/configuration/arducopter-parameters/#gnss_system_configuration_gps_gnss_mode ie trying it with a value of 3 to only fly with GPS and SBAS. A reboot is required after changing the parameter, but it doesn't make any permanent changes to the GPS so simply reverting it back to 0 will restore the previous behavior.

ThorstenDP, 2015-08-04:
Hi all,
based on some intensive tests I am pretty sure that the problem is either a hardware (GPS battery) issue or a wrong settings issue. I some cases the settings were not saved or got lost. If the settings are correct all is fine. And as mentioned by @rmackay9 and @WickedShell with rc8++ it should be solved.
From several hundreds of hours of successful flights with the M8 I think we can close this issue.

rmackay9, 2015-08-04:
Ok, great.  Closing!

closed 2015-08-04 by rmackay9

rmackay9, 2015-08-05:
@WickedShell,
Could you help me get Paul Atkin's M8 GPS running at 5hz?  It's discussed on page 221 of the AC3.3-beta testing thread.  http://diydrones.com/forum/topics/copter-3-3-beta-testing?commentId=705844%3AComment%3A2063084&xg_source=msg_com_forum