ArduPilot/ardupilot #2107: use MAV_STATE in HEARTBEAT as master caution

tridge, 2015-04-14:
This will be used to communicate critical subsystem errors. It should also be used to show that the system is calibrating

rmackay9, 2015-04-14:
copter in particular should set the state to calibrating during gyro and baro calibration.

lvale, 2015-04-16:
And what to do with these ??

libraries/GCS_MAVLink/GCS_MAVLink.h:139:    SEVERITY_LOW=1,	
libraries/GCS_MAVLink/GCS_MAVLink.h:140:    SEVERITY_MEDIUM,	
libraries/GCS_MAVLink/GCS_MAVLink.h:141:    SEVERITY_HIGH,	
libraries/GCS_MAVLink/GCS_MAVLink.h:142:    SEVERITY_CRITICAL,	
libraries/GCS_MAVLink/GCS_MAVLink.h:143:    SEVERITY_USER_RESPONSE	
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:445:	MAV_SEVERITY_EMERGENCY=0, /* System is unusable. This is a "panic" condition. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:446:	MAV_SEVERITY_ALERT=1, /* Action should be taken immediately. Indicates error in non-critical systems. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:447:	MAV_SEVERITY_CRITICAL=2, /* Action must be taken immediately. Indicates failure in a primary system. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:448:	MAV_SEVERITY_ERROR=3, /* Indicates an error in secondary/redundant systems. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:449:	MAV_SEVERITY_WARNING=4, /* Indicates about a possible future error if this is not resolved within a given timeframe. Example would be a low battery warning. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:450:	MAV_SEVERITY_NOTICE=5, /* An unusual event has occured, though not an error condition. This should be investigated for the root cause. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:451:	MAV_SEVERITY_INFO=6, /* Normal operational messages. Useful for logging. No action is required for these messages. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:452:	MAV_SEVERITY_DEBUG=7, /* Useful non-operational messages that can assist in debugging. These should not occur during normal operation. | */
libraries/GCS_MAVLink/include/mavlink/v1.0/common/common.h:453:	MAV_SEVERITY_ENUM_END=8, /*  | */

WickedShell, 2016-03-15:
Removed plane as plane does this, and marks off critical when in failsafe, emergency if it's crashed, active if it thinks its flying, standby if it thinks its just on the ground. (And calibrating goes out during init)

Naterater, 2019-01-07:
Is there a chance this might be done already? Pretty old issue here.

rmackay9, 2019-01-08:
@Naterater, it's almost done in copter.  Every failsafe except the crash detector causes it to go into CRITICAL state.  It doesn't go into initialising during startup .. it could, it's not a big change I think.

WickedShell, 2020-12-08:
I'm going to close this as done at this point. To add crash detect to copter would require persisting the crash check state while disarmed, which currently isn't done, (and if you were to have crashed and checked that it's okay you wouldn't want to still be flagging this state if you were going to take off again). Initializing isn't a mode on copter the way it is on plane, and I don't think the state is that important here.

closed 2020-12-08 by WickedShell