ArduPilot/ardupilot #2109: SITL Vagrant does not come up cleanly

hamishwillee, 2015-04-14:
As per #2096, the commit [Support building of Pixhawk firmware under Vagrant/VirtualBox](https://github.com/diydrones/ardupilot/commit/33af275220a1dcfb553ed67b8f54a273acdc85e6#diff-23b6f443c01ea2efcb4f36eedfea9089) changed the requirements for using Vagrant so the instructions in [/vagran/README.md](https://github.com/diydrones/ardupilot/blob/master/Tools/vagrant/README.md) and [Setting up SITL using Vagrant](http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/setting-up-sitl-using-vagrant/) (wiki) no longer work.

I have installed rsync and the new additional github dependencies. "Vagrant up" freezes  at certain points but can eventually be made to work. Would be good if it could come up cleanly every time.

1. It froze halfway through "vagrant up" in step "default: Cleaning up...". 
I forced it to continue with Ctrl+C (it continued for a bit then exited).
  - 1b It then crashed when  I tried to do: vagrant ssh -c "sim_vehicle.sh -j 2"

2. After doing "vagrant up" and then: vagrant ssh -c "sim_vehicle.sh -j 2" SITL worked fine.

The installation should run cleanly and not require "poking" to get it to work.

andyp1per, 2015-05-08:
At what points does it freeze? What version of virtualbox are you using? What is the host OS?

hamishwillee, 2015-05-08:
Windows 8, Virtual Box 4.3.22.

The only fail information I have is in the comment - sorry, its been a while so I cleaned up my experimentation and moved on.

magicrub, 2015-05-31:
is this still a valid issue?

andyp1per, 2015-05-31:
Without more info I think you should close. I have not seen this.

hamishwillee, 2015-05-31:
If no one else has seen it then feel free to close. Could have been a glitch on my computer.

closed 2015-06-10 by hamishwillee