ArduPilot/ardupilot #2114: How to get heading from compass as an earth-frame angle

deb0ch, 2015-04-14:
Hey guys,

Though I've been browsing all of the AP_Compass, AHRS, AttitudeControl, Navigation, NavEKF classes (and others), I could not figure a simple way of getting the drone's current heading information.

I'm developping some custom feature on the ArduCopter code and need to regulate my drone's heading (yaw) (mainly) based on compass. I need the drone to keep facing a precise direction.

The best for me would be an earth-frame angle, or a way to calculate it. I know that it is possible, since this information is displayed in the primary flight display of apmplanner2.

Thanks in advance for your help, I hope that my question is clear enough and see you soon !

rmackay9, 2015-04-14:
Questions like this should probably be sent to drones-discuss.  This issue list is really for recording bugs and enhancement requests.
The ahrs.yaw_sensor value will return the vehicle's current heading.  It's difficult to find just the compass's heading because the 3D vector is integrated into DCM and EKF.  Stlil, I believe the compass library has a get_heading method.

closed 2015-04-14 by rmackay9

deb0ch, 2015-04-15:
Thank you so much for your help, I could have spent a hundred years in the code without finding that !
 
Actually, I was looking for a getter or a kind of accessor, I never would have suspected a public attribute, so yeah I could have spent a long time searching ^^

Thanks a lot.