ArduPilot/ardupilot #2118: Official Firmware List

billbonney, 2015-04-14:
Currently in apm_planner project we only provide easy install of the current stable and the next beta release. What we really need is a list of official stable firmware release and locations so they can be easily downloaded by the user and installed

gmorph, 2015-04-16:
Hi Bill.  How would you like the list.  Some kind of XML format with release version and filename and location on the server?  
Other suggestions welcome.
Thanks, Grant.

malcom2073, 2015-04-16:
Something I stumbled upon when I was trying to figure out if there was such a list, https://github.com/diydrones/ardupilot/releases
If you upload the hex files associated with those tags and make them actual "releases", then there's already a github api to grab that information, as well as links to the hex files on github.

peterbarker, 2017-12-19:
I think this solves the initial request: http://firmware.ap.ardupilot.org/manifest.json.gz

closed 2017-12-19 by peterbarker