ArduPilot/ardupilot #2125: Smarter CH6 tuning

mrlioon, 2015-04-15:
For in flight tuning it is often essential to change two parameters simultaneously. For example:
* Rate Roll P AND Rate Roll I.
* Rate Pitch P AND Rate Pitch I.
* Throttle Accel P Throttle Accel I (i.e. "Climb Sensitivity" from Basic Tuning).

I think, it would be great to tune both of these values simultaneously via CH6. Perhaps with two different MIN/MAX values.

Actually this is already possible for some values: Rate Roll P and Rate Pitch P (or I, or D), for example. So why not offer more of these "connected" CH6 options?

peterbarker, 2025-05-12:
This isn't a great issue as it boils down to "do things better" - there's no way this issue can be closed as "complete".

I've created https://github.com/ArduPilot/ardupilot/pull/30042 which will add the flexibility to add each of the separate tuning values onto different knobs on the transmitter, allowing for simultaneous (albeit more-manual) tuning of different values.

I intend to close this PR - if there are concrete suggestions of parameter-pairs that should be tuned on the same knob then please open fresh PRs with them nominated.

rmackay9, 2025-05-12:
Thanks for the follow-up @peterbarker and I've commented on the PR but the regular transmitter interface is very limited and I'm not sure it's worth trying to extend the RC interface with more features like this

closed 2025-07-21 by rmackay9