ArduPilot/ardupilot #2133: Support for 9 flight modes in copter and plane

Noircogi, 2015-04-16:
The current code for both copter and plane allows you to select between 6 pre-configured flight modes based upon the PWM values on a particular channel (5 or 8 normally).

This is often a limitation.

Modern radios are able to produce precise PWM values which could be tested against much smaller ranges.

Using two 3-position switches in combination, it's straight-forward to create 9 step values.  The transmitter can then use voice announcing/text display to remind the pilot which mode is selected.

Optional support of 9 mode settings would be a very nice feature addition which should be straight-forward to implement but would need support in both firmware and mission planner.

abvgeej, 2015-04-16:
I want to note that you can get a unlimited selection of modes with just a single 2-pos switch, and can fully control a quad with two.
2 3-pos switches are enough to fine tune the quad from your RC with ease

I think modern radios are good enough to put additional (aka CH7-8) functionality on CH6.

rmackay9, 2015-04-16:
Thanks for teh enhancement request.
This is actually a duplicate of this existing issue so I'll close this one
https://github.com/diydrones/ardupilot/issues/1438

closed 2015-04-16 by rmackay9