ArduPilot/ardupilot #2135: Add aircraft ID parameter (feature request)

sreid55, 2015-04-16:
Add a parameter which can be written and read by the Ground Station which holds an arbitrary string value.   This will allow GS software to support assigning unique IDs to different aircraft, and maintaining separate GS profiles for each.

The parameter would have no semantic meaning within the firmware.

For example, the GS software could recognize a particular airframe by ID, and let the user store and load a few different profiles (parameter sets) for that airframe, which may include different PIDs and battery capacities that the user has created/tuned for different batteries.

iskess, 2015-04-16:
SYSID_THISMAV
Assign it a number and Mission Planner will sort the .tlogs into different folders by number.
I use the same numbers on my RC handheld model list to help me remember.
It would be really nice if we could assign a name to each model in Mission Planner, which it would correlate to the parameter value. Then it could use this name to label the folder, even each tlog. It would also show which model you are connected to.

These requests belong in the Mission Planner site, APM already has the functionality you are requesting.

rmackay9, 2015-04-16:
I'm going to close this item if that's ok.

closed 2015-04-16 by rmackay9

sreid55, 2015-04-16:
Got it - thanks for the info guys.