ArduPilot/ardupilot #2147: Floating point exception in SITL when set_position_target_local_ned has vertical speed != 0

jmacgithu, 2015-04-20:
Hello,
I'm testing the [set_position_target_local_ned](https://pixhawk.ethz.ch/mavlink/#POSITION_TARGET_LOCAL_NED) message with the SITL for arducopter. Very often a ```ERROR: Floating point exception - aborting``` appears in the simulator window when the vertical speed in the message is different from 0. It happens both for positive and negative values. However, it doesn't happen all the times, in general when the SITL has been restarted with the -w option.

magicrub, 2015-04-20:
Sounds like an ignored warning may be biting us here...

rmackay9, 2015-04-21:
Ah, that sounds terrible. FPE can't be allowed to persist.
I've been thinking of adding a SITL test of the velocity controller so I could test it as part of that.
Basically to fix this I really need to be able to recreate it... I guess that sounds easy!

jmacgithu, 2015-04-21:
Hi,
[here](https://github.com/jmacgithu/dronekit-python/blob/Example-set_position_target_local_ned/example/guided/guided.py) you can find an example. If you change the vz speed to something different from 0, it should raise an exception.
Just launch a new sitl with -w option. Wait for the simulator to load and pass all the params, get the GPS fix and then just arm throttle...

rmackay9, 2015-04-22:
I tried to recreate this problem but was unable to make the copter simulator fail.  Although I didn't use dronekit instead I used mavproxy to send the velocity commands, the sequence was nearly identical.

Because the -w option clears the eeprom I needed to first set the FLTMODE4 parameter to "4" (=Guided), then disabling arming checks (ARMING_CHECK=0), then arm the vehicle, then issue a "takeoff 5m" command then use a new "velocity 1 1 -1" command to send the velocity request.

So, I'm having some troubles recreating the failure.  I guess maybe you could try staring the simulator with GDB running (sim_vehicle.sh's -G option) and then when it fails it should show a in one of the windows where i failed.

You're on the latest version of master of course?

jmacgithu, 2015-04-22:
This is the output from GDB:
```
Program received signal SIGFPE, Arithmetic exception.
0x0806b5cf in AC_AttitudeControl::set_throttle_out(this=0x811c900 <attitude_control>, throttle_out=41484,3906,apply_angle_boost)true, filter_cutoff=2) at /home/krutemp/workspace/ardupilot-master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp:703
703   _motors.set_throttle(get_boosted_throttle(throttle_out));
```

rmackay9, 2015-04-22:
Thanks for that.

I tried for hours to recreate this exception but was unable to.  I even tried using your dronekit script but couldn't make it fail.  You're definitely using master right?

In any case, we suspect that it's the AP_AHRS libraries update_trig() function that is running before the attitude has settled.  I've pulled in a patch from Jonathan which adds additional checks, would you be able to re-test with this patch included to see if it helps?
https://github.com/diydrones/ardupilot/commit/6a6ccb4f32e3185ec17a2216187e766f6381c894

Thanks again for the report.

jmacgithu, 2015-04-22:
Hi,
the exception comes out again. The throttle value is slighlty different: throttle_out=41438,3359 (it's a float, but comes out with a coma probaly because I'm using the italian layout).
My values for the speed are vx=x, vy=0, vz=-2, but it happens even with vz=2 (throttle_out=40602,0859). The other params are still: apply_angle_boost=true, filter_cutoff=2.
If it helps, I'm using XUbuntu32 on a Oracle VM machine with 2 cores and 1024MB of RAM.

rmackay9, 2015-04-23:
very strange.  I wish I was able to recreate it.

If you can use GDB to jump back and find out the values within get_boosted_throttle that would be great.

jmacgithu, 2015-04-23:
I'm not able to use GDB well. I used some cout to print from get_boosted_throttle. Below the output. The initials stand for the variable names in that function. Before crashing, it seems it returns from get_boosted_throttle and then puts a high value of throttle in motors_set_throttle()
```
cos_tilt: 0.999841 ahrs.cos_pitch: 0.999845 ahrs.cos_roll: 0.999997
I: 1 B: 1.00016 To: 414.406 Ab: 0
cos_tilt: 0.999841 ahrs.cos_pitch: 0.999844 ahrs.cos_roll: 0.999997
I: 1 B: 1.00016 To: 408.159 Ab: 0
cos_tilt: 0.999841 ahrs.cos_pitch: 0.999844 ahrs.cos_roll: 0.999997
I: 1 B: 1.00016 To: 505.455 Ab: 0
cos_tilt: 0.999839 ahrs.cos_pitch: 0.999843 ahrs.cos_roll: 0.999997
I: 1 B: 1.00016 To: 366.632 Ab: 0
cos_tilt: 0.99984 ahrs.cos_pitch: 0.999844 ahrs.cos_roll: 0.999996
I: 1 B: 1.00016 To: 424.917 Ab: 0
cos_tilt: 0.999839 ahrs.cos_pitch: 0.999843 ahrs.cos_roll: 0.999996
I: 1 B: 1.00016 To: 446.313 Ab: 0
cos_tilt: 0.999839 ahrs.cos_pitch: 0.999843 ahrs.cos_roll: 0.999996
I: 1 B: 1.00016 To: 392.626 Ab: 0
cos_tilt: 0.999835 ahrs.cos_pitch: 0.99984 ahrs.cos_roll: 0.999995
I: 1 B: 1.00017 To: 427.238 Ab: 0
cos_tilt: 0.999833 ahrs.cos_pitch: 0.999838 ahrs.cos_roll: 0.999995
I: 1 B: 1.00017 To: 388.659 Ab: 0
cos_tilt: 0.999834 ahrs.cos_pitch: 0.999839 ahrs.cos_roll: 0.999995
I: 1 B: 1.00017 To: 452.287 Ab: 0
cos_tilt: 0.999833 ahrs.cos_pitch: 0.999839 ahrs.cos_roll: 0.999994
I: 1 B: 1.00017 To: 355.667 Ab: 0
cos_tilt: 0.999829 ahrs.cos_pitch: 0.999836 ahrs.cos_roll: 0.999994
I: 1 B: 1.00017 To: 480.064 Ab: 0
cos_tilt: 0.999828 ahrs.cos_pitch: 0.999835 ahrs.cos_roll: 0.999993
I: 1 B: 1.00017 To: 350.074 Ab: 0
cos_tilt: 0.999829 ahrs.cos_pitch: 0.999838 ahrs.cos_roll: 0.999991
I: 1 B: 1.00017 To: 399.093 Ab: 0
cos_tilt: 0.999829 ahrs.cos_pitch: 0.999838 ahrs.cos_roll: 0.999991
I: 1 B: 1.00017 To: 433.112 Ab: 0
cos_tilt: 0.999829 ahrs.cos_pitch: 0.999837 ahrs.cos_roll: 0.999991
I: 1 B: 1.00017 To: 461.585 Ab: 0
cos_tilt: 0.999825 ahrs.cos_pitch: 0.999837 ahrs.cos_roll: 0.999988
I: 1 B: 1.00017 To: 41529.1 Ab: 7

Program received signal SIGFPE, Arithmetic exception.
0x0806b755 in AC_AttitudeControl::set_throttle_out (
    this=0x811c9a0 <attitude_control>, throttle_out=41521,832, 
    apply_angle_boost=true, filter_cutoff=2)
    at /home/fabio/workspace/ardupilot-master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp:706
706             _motors.set_throttle(x);
(gdb)
```

jmacgithu, 2015-04-23:
Hi,
I think there is an overflow problem from float (throttle_out) to int16_t in:
```void                set_throttle(int16_t throttle_in)```
of AP_Motors/AP_Motors_Class.h. Any value smaller or larger than −32,768 to 32,767 causes problems.
For some reason, when a vz is set != from the previous one, there is a peak in the throttle value received from AC_AttitudeControl::set_throttle_out. The value of throttle_out it receives is directly passed to AC_AttitudeControl::get_boosted_throttle, which, from its signature, requires values from 0 to 1000.
```// returns a throttle including compensation for roll/pitch angle
// throttle value should be 0 ~ 1000
float AC_AttitudeControl::get_boosted_throttle(float throttle_in)
```
In get_boosted_throttle, the throttle value does not change much and its result is passed directly back to _motors.set_throttle. When it's really high (around 40K) and overflows the unit16_t type, the exception is raised.

I solved this problem adding a sanity check in AC_AttitudeControl::set_throttle_out, but I don't know if it solves the problem of peak throttle out values.
```
    if(throttle_out > 1000){
        throttle_out = 1000;
    }else if(throttle_out < 0){
        throttle_out = 0;
    }
```
Here is the ouput when thottle_out is negative (due to a vz positive):

```
I: 1 B: 1.01058 To: 511.903 Ab: 3
cos_tilt: 0.98947 ahrs.cos_pitch: 0.9897 ahrs.cos_roll: 0.999768
I: 1 B: 1.01064 To: 404.882 Ab: 2
cos_tilt: 0.989456 ahrs.cos_pitch: 0.989689 ahrs.cos_roll: 0.999765
I: 1 B: 1.01066 To: 483.111 Ab: 3
cos_tilt: 0.98946 ahrs.cos_pitch: 0.989696 ahrs.cos_roll: 0.999761
I: 1 B: 1.01065 To: 452.018 Ab: 3
cos_tilt: 0.989475 ahrs.cos_pitch: 0.989715 ahrs.cos_roll: 0.999757
I: 1 B: 1.01064 To: -82401.6 Ab: -868

Program received signal SIGFPE, Arithmetic exception.
0x0806b76e in AC_AttitudeControl::set_throttle_out (
    this=0x811c9a0 <attitude_control>, throttle_out=-81532,8984, 
    apply_angle_boost=true, filter_cutoff=2)
    at /home/fabio/workspace/ardupilot-master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp:709
709             _motors.set_throttle(x);
(gdb) ```

rmackay9, 2015-04-23:
Yes, I think you're right.  We're trying to mash a float into an int16.  The constraint will definitely fix the immediate problem (the floating point exception itself).  I'd like to look into it a bit more why it's coming up with such a large jump in throttle but one way or the other we will add that constrain.
Thanks!

rmackay9, 2015-04-24:
@jmacgithu,
Can you check if these two patches resolve the issue you're seeing?
https://github.com/diydrones/ardupilot/commit/ab2532a6092d0806f348ac9d4be9887c5934ee8f
https://github.com/diydrones/ardupilot/commit/a07b322d31453485b9f7bfcf92b379aa3d9752e2

jmacgithu, 2015-04-24:
@rmackay9,
The patches seem to work. However a new error comes out during gyro init with the latest commit [a07b322](https://github.com/diydrones/ardupilot/commit/a07b322d31453485b9f7bfcf92b379aa3d9752e2):
```
Program received signal SIGFPE, Arithmetic exception.
0x0808d08e in AP_InertialSensor::wait_for_sample (this=0x8116620 <ins>)
    at /home/fabio/workspace/ardupilot-master/libraries/AP_InertialSensor/AP_InertialSensor.cpp:1144
1144        if (_next_sample_usec == 0 && _delta_time <= 0) {
(gdb) 
```
It seems strange to me that I get so many aritmetic exceptions, when others don't. What's your compiling environment? Mine is:
gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) on Ubuntu 14.04.2 LTS (in reality it's a xubuntu 32bit)

jmacgithu, 2015-04-25:
Ok... It seems I've solved the problem. It seems that 32bit compilers aren't very compatible with this project. They introduce some variability when handling floats, which is otherwise handled correctly on 64 bit systems.
In fact, so far I've been working on a xubuntu 14.04.02 32bit VM. On Friday, it started giving me the problem above with the InertialSesor. I went back checking older commits and tested till when it worked and it was [0480867](https://github.com/diydrones/ardupilot/commit/0480867de22067dd1a31f4a2129d54fd892282a6). The following commit after that didn't compile and the following one gave me the SIGFPE above, even if the AP_InertialSesor file hadn't changed.
I then tried with a new ubuntu 15.04 32 bit and I had the same problem. Also with vagrant at 32 bit.
As last resort, I tried with a xubuntu 14.04.02 64 bit, and no error came out. Not even the first one which is the topic of this thread...
Which OS do you guys use? It might be that only 64 bit machines are supported, at least with linux ubuntu...

tridge, 2015-04-25:
that is a great discovery, thanks! 
I've reproduced it by building as 32 bit on my 64 bit box. I'll fix it soon.
We definitely need to support 32 bit operation.

meee1, 2015-04-25:
this is also the problem with my Cygwin builds
Program received signal SIGFPE, Arithmetic exception.
0x00447d44 in AP_InertialSensor::wait_for_sample (this=0x4b8780 <ins>)
    at /cygdrive/c/Users/hog/Desktop/DIYDrones/ardupilot-mega/libraries/AP_InertialSensor/AP_InertialSensor.cpp:1144
1144        if (next_sample_usec == 0 && delta_time <= 0) {
(gdb) info locals
now = 7412000
(gdb) print _next_sample_usec
$1 = 0
(gdb) print _delta_time
$2 = 0

so this may help on the windows sitl port in the future as well

tridge, 2015-04-27:
I've pushed a workaround that avoids FE_INVALID on i386

closed 2015-04-27 by tridge