ArduPilot/ardupilot #2148: ArduCopter climbs to 'home alt' when do_jump is set to zero

billbonney, 2015-04-20:
I just accidentally tested a scenario on ArduCopter 3.2.1 that when the do_jump was set to 0  index (ie HOME) the copter climbs to an relative altitude = to the absolute home alt. This would look like a 'flyaway'.

This behaviour was seen in ArduCopter 3.2.1 SITL.

magicrub, 2015-04-20:
Perhaps it's going to rtl_alt?

billbonney, 2015-04-20:
No, the alt was in excess of 400m with a abs home alt of 500. RTL alt is defaulted to 15m, so unlikely ;)

rmackay9, 2015-04-21:
mission zero is defined to be the home position so a jump to that mission command isn't really valid.  We could do something weird like make it RTL but it's probably best just to ignore the command (and ideally log an error in the DF log).

rmackay9, 2015-04-21:
Fixed with this commit.  https://github.com/diydrones/ardupilot/commit/87e6452ee90f267cba4cb6dbca58b5a24e92b7c0
This will go out with AC3.3-rc2.
Thanks again for the report

closed 2015-04-21 by rmackay9