ArduPilot/ardupilot #2153: PPM Module Not working correctly with AC3.3

kd7jmy, 2015-04-22:
Hello Everyone, sorry to add more work to the list but I found a bug with AC3.3 .My setup is using the 3D Robotics PPM Module with a Pixhawk in hex "x" mode, and  AR6100 RX. After loading to 3.3, I get a set of beeps.After connecting to mission planner it shows "NO RC Receiver". The Blue light on the  PPM Module does not light up with Version 3.3. If you open the radio tab you can see that the Pixhawk is only communicating with PPM Module every other second. I just downgraded back to AC 3.2.1 and everything is working just fine. Let me know if you would like me to do any additional testing.

rmackay9, 2015-04-22:
Any chance you could provide a dataflash log?
You rebooted the pixhawk after uploading the AC3.3 firmware?

kd7jmy, 2015-04-22:
I rebooted the pixhawk many times and the PPM module many times as well. Thanks for the prompt response. I will try it again at this time. I currently do not see any logs. I didn't even try to arm the copter when the message and the beeps occurred.

rmackay9, 2015-04-22:
Ok, maybe set the logging bitmask to All+LogWhileDisarmed and the logs will begin immediately upon startup.  Txs for the report.

kd7jmy, 2015-04-22:
I will give that a shot. I'm going to try an upgrade with the module
unplugged. I will post the data log asap. Thanks

kd7jmy, 2015-04-22:
What email address would you like the log sent too.

rmackay9, 2015-04-22:
drones-discuss@googlegroup.com or just rmackay9@yahoo.com.

kd7jmy, 2015-04-22:
I just emailed you the log file. Let me know what you find out.

OXINARF, 2016-11-03:
@kd7jmy Is this still valid with the latest 3.4.1 release?

kd7jmy, 2016-11-03:
There was an issue when powering from USB. When the unit is self powered
the ppm module works just fine. I believe it's a limitation of the USB.
Either the cable or the computer was the cause of the issues.

closed 2016-11-03 by OXINARF