ArduPilot/ardupilot #2155: use max current instead of max throttle for low batteries

magicrub, 2015-04-22:
battery voltage changes over time, if we limit throttle then power can reduce significantly at the end of the flight which can be problematic if the plane needs to climb for some reason.

magicrub, 2015-05-31:
having a CAN based ESC that can report the motor current would be helpful

magicrub, 2016-01-07:
The work that Event38 did does this, we will "get it for free" as soon as I do the work to integrate it in since they've stopped working on bringing it to master.
https://github.com/Event38/ardupilot/blob/E386_Autolanding3/ArduPlane/Attitude.cpp#L770

iskess, 2016-01-08:
This is key for is using Li-ion batteries that have a usable voltage range of 4.2v to <3.0v. Maybe use watts?

Thank you!

magicrub, 2016-01-08:
How about we map Watts to throttle during takeoff to find the scale and then apply that scale to an offset that is added to the throttle during the flight? That way we always have equivalent thrust for the same demanded throttle anytime during the flight, specifically at the end.

iskess, 2016-01-08:
I like that idea of mapping wattage to throttle. 
Here is a thought I've been having. I'm planning on switching from a 3S Lipo to a 4S Li-ion so I have ample power at the end of the flight if I need to execute an abort. The 3S Li-ion lands at about 9v, while the 4S Li-ion lands at 12v. 
However, the 4S at 16.8V will give me too many RPMs for my SloFly prop when taking off. So I would like to use a reduced max throttle setting for take-off. The problem is that I will need 100% throttle for an abort. Mapping the wattage to throttle will work great for auto flight, but will it be limited by the max throttle or take-off throttle parameters?

magicrub, 2016-01-08:
FYI, there a separate max throttle paran for takeoff.

rmackay9, 2016-01-08:
As a reference, in Copter we have a feature that not many people that scales the output to the motors based on the voltage.  Not saying it's perfect or that it could be easily cut over to Plane.. but you know, maybe we find some way to share info at least.

iskess, 2016-01-08:
"FYI, there a separate max throttle paran for takeoff."

Yes, I know, but since the abort code uses this same param, it will limit the throttle too much on an empty battery. Maybe this is too much of a unique problem, but Li-ion is where all the endurance limits are being broken so it seems like a good move. 
My only concern is if there is a current sensor failure, we need a fall back to % Throttle.

magicrub, 2016-02-21:
This has been implemented as BATT_MAX_WATT in PR https://github.com/diydrones/ardupilot/pull/3573

magicrub, 2016-03-04:
implemented via https://github.com/diydrones/ardupilot/commit/ed6aa4ed17e6d2d8386964375c7224d40467bd23

closed 2016-03-04 by magicrub