ArduPilot/ardupilot #2161: APM does not pick up data from UAVCAN bus

rbachtell, 2015-04-22:
Zubax gnss on AC 3.3rc1 indication on Zubax board indicates can traffic and gps lock and CLI data confirms receiving gnss signal but APM does not respond.  Pavel from Zubax is aware as is Randy.

pavel-kirienko, 2015-04-22:
I was able to reproduce this with my setup (Pixhawk, ArduCopter v3.3rc1).

```
nsh> uavcan status
ESC actuators control groups: sub: 0 / req: 0 / fds: 2
ESC mixer: NONE
Sensor 'gnss':
RX errors: 0, receiver node id: 1

Sensor 'mag':
devname: /dev/mag
channel 0: node id 1 --> class instance 1
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'baro':
devname: /dev/baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty
```

It is clear that UAVCAN driver is functioning, but APM can't use the data for some reason.

There's at least two more users who experience the same issue with APM.

pavel-kirienko, 2015-04-22:
@hsteinhaus would you mind to have a look?

hsteinhaus, 2015-04-23:
Using the latest master, everything seems to work fine here. The Zubax module got a lock about 10 seconds after boot, 30 seconds later it was good enough for APM to accept it as 3D lock (probably due to the setting of the GPS_HDOP_GOOD parameter).

@rbachtell Did you set GPS_TYPE to 9? In contrast to serial GPS units, the presence of a UAVCAN GPS currently cannot be detected automatically.

rmackay9, 2015-04-23:
Ok, I also didn't know that param change was necessary.  I've updated the GPS_TYPE parameter descriptionso that "9" is now PX4-UAVCAN.  Hopefully that will be make it more obvious for the next person.
@rbachtell, I guess when we hear back from you that this now works (remember you'll need to reboot the pixhawk after changing he GPS_TYPE parameter) we can close this issue.

rbachtell, 2015-04-23:
To all,
	After setting GPS_TYPE to 9 the UAVCAN did indeed recognize the Zubax gnss.  Short test flight and all is well on the bus.

Thank you to everyone.
RB

rmackay9, 2015-04-24:
excellent!

closed 2015-04-24 by rmackay9

RoverHunter, 2016-09-12:
Is there any tutorial how to add a canbus device to the ardupilot project?

magicrub, 2016-09-12:
Not that I know of, but looking at existing drivers is a good start. This
is a better issue for the wiki issue tracker rather than the code base
issue tracker. Please start a new issue there and we'll see about
documenting it better. closing this issue..

proficnc, 2016-09-12:
Please however do document your journey! I hope there will be many UAVCAN devices coming, and others will appreciate your work.