ArduPilot/ardupilot #2168: ArduCopter should change LED indicators to indicate vehicle capability

John-Ingersoll, 2015-04-24:
The pilots I work with use the green/blue indicators during flight as a check to see whether they can enter an altitude hold, loiter, or fly a waypoint mission.  With the recent inclusion of the OptFlow data into the NavEKF capability, the vehicle can fly without GPS data.  The LED indicators no longer indicate this capability.

The system would present better situational awareness if the LED indicators were lit by:
Prearm:
blinking blue - no aiding available (inertial only)
slow blink green - NavEKF has converged enough to allow altitude hold.
fast blink green - NavEKF will allow position control.

During flight:
solid blue - no aiding available (inertial only) (stablize, acro, etc)
blink green - NavEKF has converged enough to allow altitude hold. (altitude hold)
solid green - NavEKF will allow position control. (loiter, posHold, auto, etc).

Naterater, 2019-01-07:
I like the way the external LED's work now, although they are not aligned with this logic. Is the current 2019 logic good enough?

IamPete1, 2022-09-04:
I'm going to close, we do support a number of LED schemes. Open to ideas for new ones of course but we need more information than there is in this issue to implement.

closed 2022-09-04 by IamPete1