ArduPilot/ardupilot #2171: An idea of adding all-in-one Safety Status Display

SkiTiSu, 2015-04-25:
At present, the Pre-Arm safety check only return errors via text messages every 30 sec or arming. We could only receive one error every time.

I'd like to add a new mavlink message to represent the safety status in 1Hz, including all checks' results. Like this:

```xml
<enum name="SAFETY_STATUS_FUNC">
    <description>For SAFETY_STATUS.</description>
    <entry value="1" name="PREARM_STATUS_FUNC_RC">
        <description>RC not calibrated</description>
    </entry>
    <entry value="2" name="PREARM_STATUS_FUNC_BARO">
        <description>Baro not healthy</description>
    </entry>
    <entry value="4" name="PREARM_STATUS_FUNC_ALT">
        <description>Alt disparity</description>
    </entry>
    <entry value="8" name="PREARM_STATUS_FUNC_COMPASS_HEALTH">
        <description>Compass not healthy</description>
    </entry>
    <entry value="16" name="PREARM_STATUS_FUNC_COMPASS_CALIBRATED">
        <description>Compass not calibrated</description>
    </entry>
    <entry value="32" name="PREARM_STATUS_FUNC_COMPASS_OFFSETS">
        <description>Compass offsets too high</description>
    </entry>
    <entry value="64" name="PREARM_STATUS_FUNC_MAG_FIELD">
        <description>Check mag field</description>
    </entry>
    <entry value="128" name="PREARM_STATUS_FUNC_COMPASS_INCOSISTENT">
        <description>compasses inconsistent</description>
    </entry>
    <entry value="256" name="PREARM_STATUS_FUNC_FENCE">
        <description>check fence</description>
    </entry>
    <entry value="512" name="PREARM_STATUS_FUNC_INS">
        <description>INS not calibrated</description>
    </entry>
    <entry value="1024" name="PREARM_STATUS_FUNC_ACCELS_HEALTHY">
        <description>Accels not healthy</description>
    </entry>
    <entry value="2048" name="PREARM_STATUS_FUNC_ACCELS_INCONSISTENT">
        <description>Accels inconsistent</description>
    </entry>
    <entry value="4096" name="PREARM_STATUS_FUNC_GYRO_HEALTHY">
        <description>Gyros not healthy</description>
    </entry>
    <entry value="8192" name="PREARM_STATUS_FUNC_GYRO_CAL">
        <description>Gyro cal failed</description>
    </entry>
    <entry value="16384" name="PREARM_STATUS_FUNC_GYRO_INCONSISTENT">
        <description>Gyros inconsistent</description>
    </entry>
    <entry value="32768" name="PREARM_STATUS_FUNC_BOARD_VOLTAGE">
        <description>Check Board Voltage</description>
    </entry>
    and more....
</enum>

<message id="226" name="SAFETY_STATUS">
   <description>Add soon.</description>
   <field type="uint32_t" name="prearm">Add soon.</field>
</message> 
```

And in ```static bool pre_arm_checks(bool display_failure)```, change them like this

```cpp
pre_arm_rc_checks();
    if(!ap.pre_arm_rc_check) {
//      if (display_failure) {
//          gcs_send_text_P(SEVERITY_HIGH,PSTR("PreArm: RC not calibrated"));
        safetystatus += SAFETY_STATUS_FUNC.PREARM_STATUS_FUNC_RC;
        }
//      return false;
        flag = false;
    }
```

In MP, add a UserControl like this:
[image]

I'm the Chinese translator for MP ,and I‘m more familiar with C#. Now I'm trying to understand the code of APM. 

May it affect the performence, or have other bad effects?

Thanks.

magicrub, 2015-04-27:
Please submit a cloned issue to the GCSes.

rmackay9, 2015-05-02:
@SkiTiSu, adding this is a great idea.  It's been discussed before but none of the existing devs on the team has gotten to it.  If you want to take a crack at it we'd be overjoyed!

SkiTiSu, 2015-05-02:
@rmackay9 @magicrub Thanks a lot. I'll try to achieve.

CodeChief, 2015-05-02:
Great and also a seed for other ideas... Whilst on the subject of summaries, we could update a mini OLED display on the quad itself (those things you get cheap for electronics projects). Just imagine that UserControl directly on top of the quad or plane before you place it for takeoff.

When adding this feature, the devs could consider more "status pages" might be available in the future (and so put the code in the right place and form for future expansion) with a concept of switching to the page with a current alert or toggling through pages every few seconds.

Is there a grouping concept in MAVLink already? Group, priority/severity and last update/alert (time) attributes are what I imagine would be enough to implement the whole thing at first guess. What I mean is rather than hard-coding a single solution, add a generic grouping/refresh feature so we can bag up properties to make more sense/contextual. Maybe it even helps make the updates more efficient over the bandwidth limited telemetry links.