ArduPilot/ardupilot #2172: Airspeed Indicator code. Analog to pressure equation is needed

donaldmh, 2015-04-26:
I am currently using the airspeed indicator with APM 2.6
I have tried to use the AP_Airspeed example to read the airspeed indicator at A0 pin.
However, it gives me constant 0. It seems the indicator cannot communicate with the APM by AP_Airspeed.
I reloaded the official APM firmware with Mission planner, the airspeed indicator acts normal.
I have also tried to use the AP_AnalogSource example. It actually reads some values. (normally is around 530, if I block either of the pressure input, it goes up or down 50). which means the AnalogSource works fine.
My main question is how to utilize the AnalogSource reading to convert the voltage value to the dynamic pressure. What is the equation converting this relationship?

I went through the airspeed.h. One sentence is max( (Reading - Offset),0 ), where the reading is within [0,1023] and the offset from Mission Planner is about 2100. The result will be always 0. 
Thank you very much for all helps

magicrub, 2015-06-10:
@tridge I think you fixed this already, right?

tridge, 2015-07-30:
I have tested the AP_Airspeed example sketch on an APM2 with analog airspeed sensor and it works fine

closed 2015-07-30 by tridge