ArduPilot/ardupilot #2176: Request: Write stm32 Serial Number to .bin file

bionicbone, 2015-04-27:
Would it be possible to capture the stm32 serial number and put this in the datalogs as a msg?

This would be useful for grouping logs from the same model and also models with similar settings into my APM Log Analyser.

Thanks
Kev

rmackay9, 2015-04-27:
I was going to say we already have this in the logs but in fact, we only have the COA (certificate of authenticity) that any board manufacturer could put on their boards.  Any way, this is an example of a COA from one user's log.  It appears on aout line 464 of a dataflash log.
 PX4v2 00210024 35324719 36343032

meee1, 2015-04-27:
that number is the pixhawk serial number
the coa would be a lot larger

bionicbone, 2015-04-27:
I think that covers it :)

All my Pixhawk logs have the same number.

A few logs i have from other users have different numbers.

APM logs do not contain the information, but thats not an issue.

@Randy, i'm not sure if you like to control the close of issues so i'll leave open, let me know for future reference.

Many Thanks
Kev

closed 2015-04-27 by magicrub