ArduPilot/ardupilot #2182: Feature suggestion: add vehicle name parameter

hsteinhaus, 2015-04-28:
I would like to propose to add a vehicle name parameter, that makes it easier to remember the vehicle that a certain log belongs to. 

It would also allow to auto-generate flight logbooks (as required in some legislations for UAVs) from dataflash logs via simple scripting.

magicrub, 2015-04-28:
Like a param that is just a string of whatever you want?

magicrub, 2015-04-28:
I like that, makes it much easier to identify logs from multiple aircraft.
Very easy to do, I love the idea.

hsteinhaus, 2015-04-28:
Exactly. Just a param to store a short alphanumeric label. I wonder that nobody else missed that yet.

magicrub, 2015-04-28:
I'll jump on that, I'll submit a PR tomorrow for it. Easy to do.

hsteinhaus, 2015-04-28:
I think the main challenge is to find a nice central location (library?) that adds it to all vehicle types without touching three/four Parameters.pde files.

magicrub, 2015-04-28:
That's my plan

rmackay9, 2015-04-28:
Duplicate so staying with tradition I will close the newer one.
https://github.com/diydrones/ardupilot/issues/1411

closed 2015-04-28 by rmackay9

AndKe, 2015-05-12:
SYSID_THISMAV  - does more or less just that ?

magicrub, 2015-05-12:
Strings are best, just need to get them implemented