ArduPilot/ardupilot #2183: AC3.3: home location is at EKF origin or last disarm location not arming location

rmackay9, 2015-04-28:
The EKF does not update it's position/location while disarmed.  This has the subtle effect of causing the ahrs-home location to be incorrect if the vehicle is moved while disarmed.

This example shows the issue:
1. put the copter at position A and let it warm up.  The EKF will capture it's origin and ahrs-home will be initially set to this EKF origin.
2. move the copter 10m away
3. arm the copter.  The copter will set the home position before arming so when asked, the EKF will return the origin location again

This results in the home always being the last place that the copter disarmed, not it's current location at arming

priseborough, 2015-04-30:
This patch that should fix this:

https://github.com/priseborough/ardupilot/commit/dc97e92c6fe5a4ec1106439efdb0dc6e513fd4f9

rmackay9, 2015-05-02:
Fixed in AC3.3-rc3

closed 2015-05-02 by rmackay9