ArduPilot/ardupilot #2186: When allowing passthrough on flaperons or ailerons

marcmerlin, 2015-04-28:
As explained in details here:
http://diydrones.com/forum/topics/mixing-aileron-washout-with-auto-turn-based-on-rc1-rc5
using flaperons and crow flaps is problematic stil (I have AC 3.2)

http://plane.ardupilot.com/wiki/fixed-wing-faq/ explains how to setup crow flaps, but I don't think things are quite setup to allow what I use for my glider in manual mode.

I have flaps as passhrough for now, so APM ignores them and that's fine since I only do manual landings. I may setup flaps in ardupilot later, but don't need that at the moment.
My ailerons are setup on channel 1 and 5 with RC5_FUNCTION=18 (aileron with input).
With that, I can setup aileron washout (raise both ailerons a bit) and all works fine until I start using an automatic mode.
Because washout raises channel 1, ardupilot thinks I'm trying to turn.
It totally ignores that channel 5 is raised by the same amount (or opposite, but that's configured in my channel setup, so ardupilot knows which way things are going_)

Could you improve function 18 so that if it detects that both ailerons are going up (or down for flaperons), it cancels that out in an auto mode?
That way, only if RC1-RC5 is non zero would the RC1 value be taken as me asking for a turn.

This would allow RC controller commanded aileron washout to work, as well as allow simple flaperons to also work (i.e. only controlled by the RC controller) without confusing the auto modes.

By the way, I did look at setting up flaperons, but 
1) I want my aircraft to keep working if the pixhawk is in failsafe mode, and it's very hard to land without crow flaps. This means using FLAP_IN_CHANNEL isn't great for me since it would not work without the ardupilot brain.
2) flaperons require setting 2 channels, one with flaperon1 and one with flaperon2, but my first aileron in channel 1 (AETH), and therefore cannot be set as a flaperon1. Unless I missed something, it's another reason why flaperons aren't great for me.

Given all this, allowing for simple aileron passthrough with differential computing as a command to turn in auto mode, would make things simple for those who need just this :)

marcmerlin, 2015-04-28:
Related bug which seems to confirm you cannot set just one flaperon, and therefore apparently you have to move your aileron away from channel 1 if you want to use flaperon1 and flaperon2, as required by the flaperon code
https://github.com/diydrones/ardupilot/issues/760
I'm not against improving flaperons or making it easier, but obviously I'd be more than happy with just having my function 18 suggestion which makes everything much simpler if you are ok with just manual control.

magicrub, 2015-05-02:
@marcmerlin  is https://github.com/diydrones/ardupilot/pull/2137 related to this?

marcmerlin, 2015-05-02:
@magicrub not sure it's related since it refers to elevons, which I don't have.
This refers to simply having ailerons that can be activated as flaperons (by the RC controller), or washout (by the RC controller again).
Unless I didn't understand the docs, there is no way to tell ardupilot not to do anything else with ailerons but turn, but not be confused by control input that looks like a turn if you only look at one aileron, but then isn't one if you look at both ailerons and notice that they're deflected the same, so the RC controller didn't command a turn afterall.

JackLaLanne, 2015-05-05:
@marcmerlin thank you for bringing this up. While attempting to setup flaperons on my glider with APM 2.5.2, I came across the same problem. Tried to document everything on the forums: http://ardupilot.com/forum/viewtopic.php?f=107&t=12214

The way I interpret the wiki -please correct me if I am wrong, aileron outputs are replaced, when flaperons are configured. So one must set an RCx_FUNCTION to flaperon1 and an RCy_FUNCTION to flaperon2. But since RC1 is hardcoded to aileron this seems difficult, unless RC1 is ignored and a different RC output is used.

So for my apm it would look like this if I setup flaperons.

aileron (disconnected) =1
elevator= 2
motor=3
rudder=4
Flaperon1 (port servo) =5
pan=6
tilt=7
Flaperon2=8 (starboard servo)

No clue if that would even work.

marcmerlin, 2015-05-05:
That was my understanding too, although I wasn't clear if I do this whether I'll still have full direct access
to my ailerons from my RC controller, and whether that control would be passthrough when I'm in manual mode or if the pixhawk fails and falls back to passing on data for just 8 RC channels.
Since I have flaps and ailerons and my 8 channels are taken, I guess I could move aileron to channel 8 and mode switch from 8 to channel 1.
It's just that last time I moved an aileron away from channel one, apmplanner or mission planner wasn't happy with that because the channel numbers were hardcoded I believe.

Feel free to update this bug if you get it to work, but generally this bug was to keep the much simpler passthrough mode and simply have ardupilot properly compute whether I'm commanding a turn or not when I'm in a non manual mode.

JackLaLanne, 2015-05-05:
"Since I have flaps and ailerons and my 8 channels are taken, I guess I could move aileron to channel 8 and mode switch from 8 to channel 1."
I have just one pair of control surfaces in the wings, which I am trying to use as flaperons. Doesn't ardupilot need to know it has been deprived of any aileron output? With that zombie output 1 it just doesn't seem right.

JackLaLanne, 2015-05-08:
Success!

I connected apm 2.5.2 outputs as follows:

1 = 0
2 = elevator servo
3 = ESC throttle
4 = rudder servo
5 = port servo 
6 = starboard servo
7 = 0
8 = 0

Inputs as mapped in 9xr transmitter:

1 = AIL
2 = ELE
3 = THR
4 = RUD
5 = 0
6 = 0
7 = ID2 (full switch)
8 = P3

Paramters:

RC5_FUNCTION=24 ("Flaperon1")
RC6_FUNCTION=25 ("Flaperon2")

FLAP_IN_CHANNEL=7 (where the P1 pot is on the tx)

FLAPERON_OUTOUT=1 ("DownDown")

With this exact configuration

*My aileron stick on the tx controls the starboard and port servo as expected for the aileron input.

*In stabilize and auto modes, starboard and port servo move as expected for ailerons

*When ID2 is switched on the TX, both port and starboard control surfaces move up evenly and smooth.

*Now moving the aileron stick controls both servos in the same aileron fashion as before, just with the zero position moved upwards on both sides.

*Also true for autopilot control in stabilize and auto modes.

Problem:

*no room left for pan and tilt servos. only ch5+ can be used, 5/6 are now flaperons, 8 is mode switch so only 7 is left.

It would be nice if output 1 could be set to flaperon1.

marcmerlin, 2015-07-07:
If someone reads jack's comment (which isn't directly related to the original feature request here), I tried using AC flaperon support, and for me the flaps went in the wrong direction (up, as in spoilers and not down as flaps). Details here: https://github.com/diydrones/ardupilot/issues/2524

magicrub, 2015-07-07:
I'm afraid that this is not getting any attention by the developers because
none of them have an aircraft equipped to test and/or fix this on :(

marcmerlin, 2015-07-07:
No worries, it is what it is. In the meantime having open bugs or feature requests is better than nothing. It lets people reference them and avoid dupes.
Either way, this is a feature request, not a bug, but considering that native AC flaperon support may not be working well, doing it from the controller makes sense.
And to be clear, this can be tested on any plane, with 4 servos, you don't need a fancy 7 channel glider :)
However, I've also seen the writing on the wall that most efforts seem to be put in copters, and not planes, which is a shame for me and people who don't want to be flying 2-3kg copters that can fall out of the sky and hurt people when they fail, but developers either choose what they'd like to work on, or in the case of 3DR, what their boss tells them to :)

marcmerlin, 2015-07-08:
Hi @JackLaLanne can you comment on #2524 whether you are getting flaps or spoilers with this setup?

marcmerlin, 2016-03-29:
Anyone reading this issue so far, I *think* you can setup flaperons going through ardupilot (i.e. not using your RC controller mixing) by moving your aileron RC output channel to 8, setup that channel as flaperon too (so that the same channel is used as aileron in and aileron out, fixing the problem of losing a valuable RC channel below the 8 mark), and then moving mode switch to channel 1.
This is totally untested though, some stuff may still have aileron hardcoded as channel 1.
Obviously allowing flaperon out on channel 1 would make things much easier.
Or supporting the original use case described here, namely computing aileron input for AP controlled modes, from a differential of 2 RC channels.

OXINARF, 2017-06-04:
@marcmerlin Is this still an issue with Plane 3.8.0 beta?

marcmerlin, 2017-06-04:
Yes, nothing has changed there as far as I know.
I've worked around the problem by programming my RC controller not to actuate flaperons in any mode other than MANUAL.
However, this is a pretty corner case thing. Fixing https://github.com/ArduPilot/ardupilot/issues/6295 would make this bug irrelevant.

obor00, 2017-07-13:
Strange that no  one has really fix this problem. It's so basic for any  serious airplane to  have aileron differnential and flap mix.  Coding a programmable mixer near servo output would  probably fix the thing. 
Aileron  at 0%  -> 0% ch 5, 0% ch 6
Aileron at +100% -> 100% ch5, (100-diff)%  ch6
Aileron at -100% -> (100-diff)% ch5, 100% ch6

marcmerlin, 2017-07-13:
@obor00 I had this chat with @magicrub saying the same thing :) and basically my feeling is that ardupilot developers do not fly bigger planes where crow flaps are kind of essential.
Ardupilot does have flaps support, and/or flaperon support, but even flaperons I'm not sure that many people use, it took me a long time to figure out how to make them work and document it:
http://marc.merlins.org/perso/rc/post_2015-08-04_Using-Flaperons-With-Ardupilot.html
alledgedly as per my doc you should be able to setup ailerons as reversed flaperons and flaps as flaps at the same time but I didn't try it due to how complex it is, how it burned an extra RC channel (maybe not anymore in 3.8), and how I'd likely be the single user on earth to be doing this with AP (i.e. test pilot with codepaths that aren't being used by anyone else in that way)

obor00, 2017-07-13:
@macmerlin, no you are not alone on earth to use aileron differential :) it s so basic. Flaperon is something else, most modern plane have 4 servos in wings, aileron and flaps with ailerons differential, crow braking ( butterfly) etc... If a developper want to develop this, i will be happy to help testing :)

marcmerlin, 2017-07-13:
You need to read my long-ish doc. My point is that you can use flaperon support to do the braking part of crow flaps. It's not meant for that, but my inverting the signal, it will try to use the ailerons as flaps by pointing them up instead of down. Then you can also have flaps signal sent to 2 actual flaps.
I'm pretty sure it can work as is, but the setup is complicated.

Naterater, 2018-12-28:
This is complete with crow flaps: https://github.com/ArduPilot/ardupilot/pull/9391

closed 2018-12-28 by Naterater