ArduPilot/ardupilot #2190: AC3.2 does not allow disarming with rudder like copter does

marcmerlin, 2015-04-28:
I have BRD_SAFETYENABLE=0 and ARMING_DIS_RUD=0 on my plane.

This is because I wanted my servos to work as soon as the pixhawk is booted, or if it were to crash or reboot in flight.
In other words, I found the arming switch totally pointless when I have ARMING_DIS_RUD=0 which is the real safety for my motor, and to be more precise, I found Arduplane less safe than it should be by default since it will still crash my plane if it reboots in flight because all servos are frozen waiting for me to push the arming switch.
So, I'm thankful for BRD_SAFETYENABLE, but would recommend making this a default if the arming by rudder is set.

Now, my main problem is that after landing, I can't disable arming without a ground station.
My requests would be
1) allow throttle down, rudder left to disable arming, like on a copter
2) if BRD_SAFETYENABLE is enabled, change the safety switch into a motor arming switch

Right now, I can push the safety switch after landing, but this enables the safety, it does not disarm the motors, which is what I would like.

magicrub, 2015-04-29:
I thought we had rudder disarm but I just checked and we don't, doh! Another divergance form copter. I think it is a good thing to have and it should get rolled into a common library for rover and copter to share code with.

tridge, 2015-07-23:
this is fixed in master with the new ARMING_RUDDER parameter

closed 2015-07-23 by tridge