ArduPilot/ardupilot #2191: APM Plane: re-set home location when safety switch pushed and motors armed?

marcmerlin, 2015-04-28:
I have EKF enabled, 
http://plane.ardupilot.com/wiki/flying/starting-up-and-calibrating-arduplane/ says I can set SKIP_GYRO_CAL = 0 after one flight, but another doc elsewhere says it's a bad idea with EKF.
Should that doc be updated?
Also, the doc doesn't make it very clear if my plane should be level on the ground or not for GYRO_CAL. the way mine is, it points nose up when it's on the ground.

But back to the main point, I basically init my plane next to my car, wait for gyros, setup my video, ground station and all, and by then my pixhawk has GPS lock and sets home location to that point.
But I don't really want this to be the point where the plane will rtl and circle since it's a no fly zone.

I understand I can setup a separate home location manually, but I was thinking: would it make sense for APM Plane to set the home location to where it first gets GPS lock, and then set it again when the safety switch is pressed (if used), and set it again when motors are armed (if used).

magicrub, 2015-04-29:
The current behavior is while disarmed the home position is constantly
being set. That sounds like it would work for you, right?

The gyro cal just needs the plans to be still, it can be in any orientation.

marcmerlin, 2015-04-29:
Thanks for the reply @magicrub 
Mmmh, interesting, I thought I saw something else, hence my opening this issue, but maybe I did something wrong.
Just to be clear, engaging the safety switch (automatic in my case) does _not_ set HOME, but arming the motors (which is pushing the safety switch for some, or using rudder arming for others), will set HOME, correct?
If so, it sounds like it's just a matter of mentioning this in the page mentioned above and/or other relevant page(s) along with a warning that SKIP_GYRO_CAL is apparently not a good idea if you are using EKF.
Does that sound right?
If I have or can get permissions, I'm happy to modify the page mentioned above.

magicrub, 2015-04-29:
I believe that is how it works, at least on recent versions. Which version
are you running?

We should have park chime in about the gyro cal. I was doing what you were
doing, doing it one, but eventually just set it to always do it. It's
temperature dependent so it's a good idea to do it all the time, doesn't
hurt.

Tridge, can you get ahold of Hamish to get Marc wiki permission?

marcmerlin, 2015-04-29:
AC 3.2.3 or whatever the latest is.
Did you mean to say @tridge to get his attention?
Obviously, before I go write on a page that everyone reads, let's make sure I'm not going to make it worse :)
We already know now that gyro cal doesn't require the aircraft to be level, just still.
Then you can have someone who knows best say whether I should indeed write that with EKF you really should do gyro cal each time you init.
Thanks

tridge, 2015-07-30:
i've updated the docs to make the behaviour of the home position clearer, and not to mention SKIP_GYRO_CAL.
We've found that the variation with temperature of gyro zero means that SKIP_GYRO_CAL is a bad idea except under special circumstances (eg. the need for in-flight reboot)

closed 2015-07-30 by tridge

marcmerlin, 2015-08-03:
Thanks @tridge much appreciated.