ArduPilot/ardupilot #2199: Copter: MOT_SPIN_ARMED only has effect after reboot, safety issue

chapman, 2015-04-28:
Changing MOT_SPIN_ARMED seems to only take effect after a reboot. This is a bit of a safety issue, e.g. turn it off to test something like pre-arming checks, expecting them not to spin and they do.

Changing this should take effect on the next arming, rather than the next arming after reboot.

rmackay9, 2015-04-29:
yes, makes sense.

jschall, 2015-04-29:
There is a larger issue that a whole bunch of our parameters have this property.

rmackay9, 2015-04-30:
A work around might be to add a "@RebootRequired: True" to the parameter description and ask the GCS developers to pop up a notifaction to the user.  That's obviously not as good a solution as making all params take effect immediately but it would be far easier to implement for the large number of parameters that have this "feature".

lthall, 2018-01-11:
Looks like Randy has sorted this one.

closed 2018-01-11 by lthall