ArduPilot/ardupilot #2201: AC3.3: set ahrs home alt to EKF origin alt or baro-alt when we get ahrs home in the air

rmackay9, 2015-04-29:
In AC3.3 we set the ahrs home only after we receive EKF origin.  This has two bad impacts currently:
1. current-alt appears as zero until the EKF origin is set (we should probably report alt-above-origin)
2. when we get EKF-origin/AHRS-home the alt-above-home is set to "0" and when we RTL, we will return to RTL_ALT above that in-air altitude.  It's very likely this will be unnecessarily high and will lead to an unnecessarily slow descent (because the vehicle will reduce it's descent to LAND_SPEED at a high altitude).
What we should probably do is when we set the home position (after getting the EKF-origin) is to set the home-alt to the EKF-origin-alt.  This is the right altitude if we assume the ground at the EKF-origin lat/lon and the ground at the ahrs-home lat/lon position are at the same physical altitude.  A more sophisticated option might be to use the EKF-origin-alt + (ahrs-home-terrain-alt - EKF-origin-terrain-alt) but that's likely more complicated than is necessary.

priseborough, 2015-05-13:
Setting home alt to EKF origin alt works if the flat terrain assumption is OK. There will be some drift in altitude between when the EKF origin is set and we fly due to normal baro drift, but unless we are using aided GPS, the GPS height uncertainty will be larger than that in most cases..

rmackay9, 2015-05-13:
Fixed with these two patches:
https://github.com/diydrones/ardupilot/commit/2a1a1054620a3cb55d0c0724c745e98bc85f59ca
https://github.com/diydrones/ardupilot/commit/1f96cbd015b8d8245dd5cfe258c2de8ff77d6d55

As Paul says it's not a perfect fix because it doesn't take account of baro drift (or the vehicle being carried to a high altitude) after power-up but before take-off.  Still adding extra complexity to capture the difference between the ekf-origin-alt and the take-off alt seemed like it was more likely to lead to errors in other parts of the code.

closed 2015-05-13 by rmackay9