ArduPilot/ardupilot #2204: compassmot not present in arduplane 3.2.3

marcmerlin, 2015-04-29:
I know I've used this on arduplane last year. Was it removed?

I went in console, typed setup, help shows "compass" but not "compassmot"
Typing compassmot says "invalid command"

I realize the copter guys care more about this, but I'm getting 
Large change in mag_field (39.95%) Max mag field length (634.77) > recommended (550.00)
on my plane with a 1000W electric motor, so I figured I'd try compassmot.

I have a pixhawk with external compass, and it's 20cm from the motor, but it's still sitting on top of a bunch of electronics and wires I can't move. I'm willing to try other things to fix this, but didn't find docs that mentioned anything else than compassmot.

No idea if it being gone in APM::Plane is a bug, or on purpose.

jschall, 2015-04-29:
It never was implemented on plane. We will eventually start learning it
automatically eventually.

For now the best you can do is try to get high current wires as far away as
you can.

marcmerlin, 2015-04-29:
@jschall I'm starting to worry about my memory, I could swear I did that on an APM 2.6 plane last year. Maybe I thought I did but it didn't actually happen.
If you agree compassmot would be useful in APM::Plane, let's leave this open. If not, feel free to close it.

magicrub, 2015-05-02:
What @jschall is trying to say is the current compassmot won't be ported from copter to plane because compassmot will soon be replaced by an in-flight auto-learning scheme for both platforms.

marcmerlin, 2015-05-02:
understood, let's close this then.

closed 2015-05-02 by marcmerlin

marcmerlin, 2016-03-26:
Eh, it's almost been a year now, does 3.5.1 support compass auto learning?
My last flight, I still got
               Compass:  FAIL       Large change in mag_field (83.06%)                         [GRAPH]

but I can't tell if it's the internal compass or external compass, and since I don't have compassmot to help, I have no idea what to do :-/