ArduPilot/ardupilot #2212: wishlist: data logs should be saved in filenames with leading zeros

marcmerlin, 2015-04-30:
Currently I get 1.BIN 9.BIN 10.BIN. 100.BIN...
This doesn't sort well in ls.
Can that be changed to 0000001.BIN 0000010.BIN etc ?

magicrub, 2015-04-30:
The date wouldn't hurt either but I'll take that battle to another issue :)

peterbarker, 2017-02-10:
@magicrub can't use date as the vehicle may not have an external time reference at all, and even when using GPS logging may start before we see a satellite.

marcmerlin, 2017-02-10:
@peterbarker true, so for now could the sprintf just be changed to have leading zeros?

peterbarker, 2017-02-10:
@marcmerlin Testing the patch now.

peterbarker, 2017-02-10:
```
>>>> PASSED STEP: fly.ArduCopter at Sat Feb 11 09:14:38 2017
check step:  fly.ArduCopter
Renaming logs/00000002.BIN to /home/pbarker/rc/buildlogs/ArduCopter-00000002.BIN
Renaming logs/00000003.BIN to /home/pbarker/rc/buildlogs/ArduCopter-00000003.BIN
Renaming logs/00000001.BIN to /home/pbarker/rc/buildlogs/ArduCopter-00000001.BIN
```
So far so good.

peterbarker, 2017-02-11:
On Fri, 10 Feb 2017, Marc MERLIN wrote:

> @peterbarker true, so for now could the sprintf just be changed to have
> leading zeros?

Turns out to be a little tricker than that; need to retain compatabiity 
with the files already present (which means that the function that hands 
you back a filename may stat() for it first.

Nothing can ever be simple.

magicrub, 2017-02-11:
Better late than never! Thanks for jumping on this Peter...

PR #5709 (2017-02-14): DataFlash: zero-pad the number in the log filename
This helps with collating when reviewing log files

Closes #2212
commit 057819cc (2017-02-14): DataFlash: zero-pad the number in the log filename

This helps with collating when reviewing log files

Closes #2212
closed 2017-02-14 by tridge