ArduPilot/ardupilot #2213: wishlist: speak wav files from sd instead of using beep codes

marcmerlin, 2015-04-30:
Pixhawk and newer platforms are powerful enough to play a wav file from SD.
My $200 RC Taranis controller does this.

APM:Plane and Copter have way too many beep codes and flashing colors of LEDs for anyone to reasonably remember, and you don't want to pull your decoder ring when you hear a beep sequence in flight.

Even if this may not be possible on older APM 2.6 platforms, newer ones should be able to play wav files from the sdcard, and I'm ok with the audio being choppy or not easy to hear on occasion if the CPU is too busy doing more important things like flying the aircraft :)

This would be an option for those who want it, and beep codes would still be played if the relevant wav file for each beep code, isn't on the sdcard.

rmackay9, 2015-05-01:
Great idea.  It's been discussed before and the feedback from the px4 guys was that the Pixhawk's speaker isn't up to this task and the best we would get is to make is sound like those 80s video games.
http://en.wikipedia.org/wiki/RealSound

marcmerlin, 2015-05-01:
@rmackay9 thanks for the answer. 
I had computers in the 80's :) and they sure could do sound and speech well enough, so I'm not sure you're saying the pixhawk itself doesn't have the horsepower to do this.
So to be clear, is it just a speaker problem? I'd happily upgrade it and I'm sure 3DR would be happy to sell upgrades :)
Also, as-is realsound-like quality would still work better for me than a beep code I just can't recognize.

rmackay9, 2015-05-01:
@marcmerlin, 
3DR is so busy with the Solo I don't think we can count on them to create this upgrade.  That's not all bad news though, it's an opportunity for someone else to step in and make it.
So if someone comes up with a commercially available device that can do speech and be attached to the Pixhawk and then sends a couple of those devices to members of the dev team one of us can add support.

marcmerlin, 2015-05-01:
Just to make sure I understand you:
Are you saying the problem is not the buzzer, but the fact that the pixhawk is unable to put out sound quickly enough to it, to play wavs or some other digitized audio in a way that you can understand human speech?
So were you referring to a 3rd party standalone device that would receive audio and then send it to a DAC that is better than what's on the pixhawk?

rmackay9, 2015-05-01:
What I understood from the px4 guys (who designed the pixhawk) is that the limitation for doing voice on the pixhawk is the buzzer.  That physical device won't allow proper speech.

marcmerlin, 2015-05-01:
Thanks for confirming. In that case, if it's just a matter of finding a small speaker that can be plugged in its place, this shouldn't be too hard to find.
Would you like me to buy a few samples and mail them to you?

rmackay9, 2015-05-01:
@marcmerlin, Sure.  Or if you find one from some retailer like Sparkfun I can order direct.  Either is ok.

prickle, 2015-05-02:
I also think this is a great idea. I have had very legible and surprisingly loud results for speech from a small mylar speaker, like http://www.mcmelectronics.com/product/55-4504 when baffled reasonably well.

marcmerlin, 2015-05-02:
@prickle I checked that speaker, it's apparently 13g, which isn't too unreasonable.
@rmackay9 would you like me to send you one?

More generally, I tried plugging the pixhawk buzzer in my laptop to listen to a movie. As you warned, the sound is not great, but speech is mostly understandable.
My guess is that by carefully choosing the wav samples, pixhawk could alternate the beep code (likely louder from a distance) and the voice which will be understandable on the bench or when setting up before flight, but not in the air. Sure the voice is scratchy, but the point is to just understand the message, and for people who can't understand it, they still have the beep code that alternates.
Once the support is added, I'm certain people will find lots of small speaker upgrades if they'd like louder/better voice quality.

rmackay9, 2015-05-04:
@marcmerlin, the computer test is interesting and indicates that I was wrong about the physical buzzer being the issue.  So, that means the issue is that the output pin from the Pixhawk (that connects to the buzzer) is the wrong type.  It can only send pwm output which results in different tones but not speech.  Sorry, about that.  You know, instead of me guessing anymore, I'm going to ask Lorenz from the px4 group to weigh in here.

rmackay9, 2015-05-04:
I checked with Lorenz and he says we'd need a voice synthesizer chip or "smart phone voice out".

magicrub, 2015-05-04:
A pwm will work, that's how a class D amplifier does it. We just need to
add an external Low pass filter to remove the square wave harmonics. Simple
RC circuit would do it.

CodeChief, 2015-05-04:
Hobby King just added e an RC sound system and speakers built for the task even to be heard at a distance. What's the limitation/capability of Pixhawk in this area? PWM seems to be the golden workaround for a number of features, but doesn't all this software modulation hit the CPU? As you have a taranis best get the FrSky telemetry option then you have most speech directly, and/or an Android tablet/phone attached to the transmitter will speak all the same APM specific messages directly (sort of if you ignore the radio link inbetween). Personally I hate the buzzer and would prefer one half as loud. I added a micro DIP switch to mine so I can work on it at home without waking my family late at night! In the long term speaker support would be cool. Besides a cheaper RC sound system for sound effects (the HK one is overpriced in my opinion) there could be practical applications like drines distribiting warning messages and critical information in disaster areas.

marcmerlin, 2015-05-04:
@CodeChief do you hve  link ot the hobbyking system?
Taranis telemetry would be good, but I use an L9R (long range without telemetry), also this is mostly about the many beep codes I get from pixhawk before I start flying or on the bench, which taranis can't help with.
Android tablet, yes, I would totally do this if it worked for me, but it doesn't. I can send commands from my phone/tablet, but not receive any telemetry back. No idea what the bug is: http://ardupilot.com/forum/viewtopic.php?f=15&t=12171&p=32603#p32603
Also, keep in mind not everyone has 3DR radios in their model (my small plane, I can't fit one), and I try not to use it when I work at home because it messes up all my weather sensors and other home automation running on the same frequency :)

As for speaking audio to the ground while in flight, that sounds fantastic, but for me that's a stretch goal (volume being mostly an issue and potentially requiring an amplified system with volume control and that would likely be outside of the flight controller)
And adding an external pot to tone down the volume of the buzzer when working at home without wanting to wake up family, that's a good idea, but obviously we can all already do this without code changes.

CodeChief, 2015-05-04:
The link to the sound system is here:

http://www.hobbyking.com/hobbyking/store/__78418__MrRC_Sound_V4_1_Speakerless_Sound_System.html

They also have a micro telemetry radio which may fit in your plane.

https://www.hobbyking.com/hobbyking/store/uh_viewItem.asp?idproduct=74647

And even one with integrated OSD (saves another device and serial port/y-cable if you have FPV).

https://www.hobbyking.com/hobbyking/store/uh_viewItem.asp?idproduct=74650

I presume you configured the transmit power in the 3dr radio flash/configuration tool (separate download from APM site, the internal tool appears to be broken for a while now)? Check the radio frequency ranges, number of hopping channels and transmit power against the table documented on the APM site to try and avoid interference.

That MrRC system uses your plane wings as the speaker so you should hear it. Maybe the electronics board works with a loudspeaker too. It's an interesting solution, shame it's a bit expensive just to test and the HK video does not show it working.

On further examination it won't be any good for this requirement. For APM we need a device we can control, this one requires new chips for different sounds and is hardwired to trigger manually via RC transmitter input. Some of the ideas and maybe the speaker could be reused though.

marcmerlin, 2015-05-04:
@CodeChief very cool that radio+OSD integrated
My 3DR radio should be configured ok, because it works fine if I put the same receiver into my laptop. When I then move it to my phone, my phone can send, but not receive. Can discuss further on the post I gave instead of this bug if you wish.

As for the sound system you linked, looks beefy but indeed $150 plus the weight requirement are a bit steep.
The earlier suggestion of using a PPM channel with maybe a resistor and capacitor to round the square waves, and plugging that into a good buzzer, should work as is on the pixhawk.
Is this something worth pursuing? (given someone with the skills and time of course)

abvgeej, 2015-06-29:
If not voice, then at least some support for XM playback :)

UT99310, 2015-07-26:
Somewhat related, I'm looking at the code changes necessary to use an AUX pin for a piezo speaker/buzzer (for the signal tones) instead of the SPKR pin, for those boards that do not have SPKR available. If anyone else is working on this, or might want to coordinate with me on this, please let me know. Thanks!

IamPete1, 2021-08-30:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/8884

closed 2021-08-30 by IamPete1