ArduPilot/ardupilot #2214: Use general handler for MAV_CMD_NAV_RETURN_TO_LAUNCH in Rover

hamishwillee, 2015-04-30:
Rover has a specific handler for MAV_CMD_NAV_RETURN_TO_LAUNCH in [commands_logic.pde](https://github.com/diydrones/ardupilot/blob/master/APMrover2/commands_logic.pde#L41) but does not do any specific handling. This gives the false impression that this method is implemented by casual inspection of commands_logic.pde. If would be better if this was just handled by the general/default handler.

OXINARF, 2017-06-02:
I'm not sure what this issue means. The commands_logic file handles the commands inside a mission. Unless I'm forgetting something there is no general handler for missions, each vehicle does its own handling.

The code from when this issue was opened looks to be the same as it is now so I'm not yet closing as fixed.

@hamishwillee can you elaborate more here?

hamishwillee, 2017-06-02:
@OXINARF  This is pretty old, but as I recall I was documenting the supported commands in Rover and this one had an empty method. The implication of having the command handler is that the method is supported, which I found confusing. So my suggestion was to remove the method or otherwise make it clear in code that Rover does not RTL when it gets this command in a mission.

I could of course be wrong, and things could have changed.

OXINARF, 2017-06-02:
Thanks for trying to explain it. I've looked again and I've went as far as the beginning of APMrover2 and RTL was always defined with a method with content. Maybe you saw it wrongly :slightly_smiling_face: 

Anyway, I'll close this as it doesn't seem to be valid.

P.S.: Sorry for all the SPAM to you today, I'm going through open issues from team members.

closed 2017-06-02 by OXINARF

hamishwillee, 2017-06-02:
I'm very happy to be wrong. Awesome to see you able to find the time to do this work (I know you receive a little pay, but nothing compared to the effort). It is essential.