ArduPilot/ardupilot #2216: Feature Request: Antenna Tracker - Enable tracking of a fixed location

ryanjohnston4, 2015-05-01:
It would be great if we could set a fixed location to be tracked for a movable or airborne antenna tracker. The antenna tracker would then not need anything but GPS and the data from its own sensors and the predefined location from its initial power up (or if you set it otherwise). No telemetry/mavlink connection is needed and the tracker can be used in the air with its sensors to point a small gimbal with the desired antennas to the specific location on the ground.

rmackay9, 2015-05-01:
Ok I have been thinking of adding support for the do-set-roi command which would fit with this.  It would then always point at that location until it's mode was changed or it received another command.

If the intention is to use this on a flying vehicle to make it always point at the home location then it might be better to instead use the gimbal to do that.  If the issue is that copter/plane only support one gimbal then that can be fixed (in fact, the recent front-end/backend split for mount makes it easy).  We could also add a mode to the AP_Mount library which just forces it to point at the vehicle's home position so there wouldn' be a need to send a do-set-roi command to the vehicle or mount to make it point at home.

ryanjohnston4, 2015-05-01:
Thanks Randy - do-set-roi would seem like a good fit but would it be able to take altitude into account? Also, would be good to just lock it in that mode if wanted so it would be independent of a telemetry connection or if the connection simply dropped out.

Regarding the gimbal solution and using mount-control or something supported (storm32, etc), it would probably be overkill or more difficult to use because of the brushless setup which isn’t needed for an antenna at distance. Pan/Tilt servo setup from antenna tracker on an independent APM would be the smallest and completely independent. Set it and forget it to track the home location or power on would be hassle free.