ArduPilot/ardupilot #2218: wishlist: Allow data logging of motor RPM and/or frsky sbus sensors

marcmerlin, 2015-05-01:
I saw with @tridge and a bunch of 3DR folks at the recent ELC that thre was support being added for a quadcopter ESC that reports RPM data to the flight controller.
However, it's only a 20A ESC, which is entirely useless for my high power plane which uses 80A :)

Castle ESCs already report RPM on a 4th white wire, and can actually report a lot of information as explained here: http://www.castlecreations.com/support/documents/castle_link_live_2.0.pdf (page 6)

For people with any other ESC, you can just add this $18 sensor: 
http://www.alofthobbies.com/frsky-sp-rpm.html and you even get 2 free temperature logs, like how hot your ESC and motor are getting.

It would be awesome for APM to add support for widgets like this and add it to the data logging.
Adding that generic support would even allow adding RPM support for quadcopters without replacing their ESCs,or using their existing ESC if it reports RPM already.

CodeChief, 2015-05-01:
I heard this feature is being worked on indirectly by people currently trying to get generic support for I2C into APM. It's mixed in with the requirement for faster ESC update, such as OneShot125 and other alternative interconnects like CAN.

Why indirectly? Because if you read this https://github.com/sim-/tgy/issues/64 and use an I2C variant of the SimonK firmware like this one adapted by balrogk1n of RC Groups here http://www.rcgroups.com/forums/showthread.php?t=2032252 you will get both RPM and temperature without any additional sensors!

The video posted by him here https://vimeo.com/78793047 is impressive. It looks like exactly what we want. The best part is we should also get the high speed update together with all the additional information. From the comments there I gather each ESC manufacturer and model has it's own capabilities, is more or less difficult.

Such complexities need to be hidden away from the user. That's a reason why I suggested some kind of GUI selectable profile (in an earlier discussion about OneShot125 and I2C) when implementing smarter ESC interconnects in APM. The community could help build the profiles, so we get a nice drop-down list of almost every common ESC model for the average user to just select then get the full capabilities from their ESCs.

So to APM devs please expedite the re-factoring of the ESC code (last I heard some kind of more generic splitting of the code was required) and I2C support, because we get a back-channel that will enable a load of other features together with it!

Besides being good to look at and log for problem diagnosis, I could imagine some kind of smart motor control like overspeed and overheat prevention, predictive motor failure detection (emergency landings) and overall performance/optimization indication (e.g. detection of sync issues). How cool would all that be?! :-)

marcmerlin, 2015-05-01:
@CodeChief this does all sound great indeed.
I agree that supporting multiple kinds of ESC feedback systems would be great. The SimonK firmware is all nice and great, but not useful for my Castle 80A ESCs :)
Add adding support for http://www.alofthobbies.com/frsky-sp-rpm.html  would easily give access to monitoring for everyone since it's an add on sensor you can add to any ESC (plus I love the idea of being to monitor both my ESC temperature and the motor temperature with it).

WickedShell, 2016-03-15:
Spektrum telemetry protocol for their collection of sensors: http://www.rcgroups.com/forums/showthread.php?t=1726960&page=22#post34091390

marcmerlin, 2016-04-03:
Anyone following this issue, AP 3.5 has RPM logging, although I haven't gotten it to work on pixhawk yet, maybe others have: http://discuss.ardupilot.org/t/ap3-5-rpm-not-returning-anything/7892

That said, turns out I need temperature logging even more badly than RPM logging as for a plane, you can have high power motors with insufficient cooling, that can only be run for some amount of seconds at full throttle. Guessing has caused me to burn 2 motors already (yes, using a bigger motor with more cooling area would help, but there is no room for it in my motor glider).
At this point, being able to measure/log one temperature, ideally 2, would be actually even more helpful than RPM logging.
Once that's in AP, I should be able to hack the minimosd code to add support for displaying that.

lzl200102109, 2017-07-15:
Hi, I am using ArduCopter v3.3.3 right now, and I would like to obtain RPM readings for the motors. I saw that there are several ESC messages defined in the firmware in the data log code. They include motor RPM as one of the logging variables. But my IRIS+ does not log these ESC messages currently. I was wondering what kind of ESCs could enable the RPM logging for these ESC messages.

marcmerlin, 2017-07-15:
@lzl200102109  issues are not the right place for user support. Please use https://discuss.ardupilot.org/

gotmynina, 2018-09-04:
Is this still in the works? I think support for Castle Live link would be very useful to for rover applications, It can send rpm, battery voltage/current , bec voltage/current, electrical RPM, temp, etc., and take throtte input all on one wire!, with castle escs that have an aux wire even more options are available

IamPete1, 2022-09-04:
We now support logging a bunch of ESC protocols.

closed 2022-09-04 by IamPete1