ArduPilot/ardupilot #2221: Wishlist: APM Plane: Shake On Shake Off

bchristal, 2015-05-01:
Shake the plane to start or stop the motor during auto takeoff. Same method used by eBee.

magicrub, 2015-05-01:
Can you walk us through the launching process? Is there a video of eBee
doing this?

bchristal, 2015-05-01:
Here's a video: https://www.youtube.com/watch?v=oZ3ZSnQM7v4 see around 1:45 for the shake start. This is how I imagine it working:
Create waypoint mission with Auto Take off (set to shake)
Start the mission (transmitter switch or Mission Planner or Droid Planner)
Pick up "live" airplane
Shake in a distinguishing pattern 
Motor starts. You've got a good grip on the airplane since you were just shaking it.
Hand launch and start mission OR shake again to stop motor and pause mission. This isn't shown on the video, but it's there. Think about it. You need a way to stop the motor - especially if you aren't using a transmitter.

-Brian

magicrub, 2015-05-02:
I like it, it could even be simply added as a param to the TAKEOFF mission command

rsesa, 2015-06-15:
isn't it already working by setting a non-zero TKOFF_THR_MINACC and an auto-takeoff?

magicrub, 2015-06-15:
@rsesa  Yes and no, with TKOFF_THR_MINACC > 0 it must detect the given -(accel.X) value once. In the shake on method it would be a +X, -X, +X, -X sequence within a certain timespan. Very doable, perhaps we could signify this mode using a negative TKOFF_THR_MINACC value?

rsesa, 2015-06-15:
Could work! Since then I'll use this method and pretend I have an eBee, no one will know!

But the eBee's method is definitelly safer.

bchristal, 2015-06-15:
LOL. Fantastic. I'm cutting flying wings out of black foam now! How about shake off?

jcooper21, 2016-10-24:
Any movement on this?  Its a fairly important safety feature for hand launching planes with push props.  Both sensefly and PX4 use it for that reason.

gmorph, 2016-10-24:
I use the shake method all the time with Ardupilot to turn the prop on.  Works great.  You could "add" it as a feature but its working very well right now so I don't see that we gain anything.
Thanks, Grant.

jcooper21, 2016-10-24:
@gmorph how did you enable shake on for mission flights with APM?

gmorph, 2016-10-24:
I have an auto take mission setup and TKOFF_THR_MINACC set to 4 so it detects when I shake the plane.  When I'm ready I pickup the plane, carry it to where I want to takeoff, I flick the switch on my transmitter to put the plane into AUTO, shake it until the prop starts and then throw it.
Thanks, Grant.

magicrub, 2016-10-24:
@gmorph There's a big saftey difference between when you're doing /w TKOFF_THR_MINACC and what @jcooper21 is asking for that eBee does. A single bump will trigger the throttle, @jcooper21 is looking for a deliberate trigger via multiple pumps

bchristal, 2016-10-24:
Don't forget shake off as well. It's kind of hard to put the plane down when it's wide open if you need to abort.

magicrub, 2016-10-24:
as of right now, if you do an TKOFF_THR_MINACC based takeoff but the accel trigger is not from it being thrown (ie, shaking it in your hand) and with CRASH_DETECT>0 then you'll instantly "crash" and the motor will disarm quickly

jcooper21, 2016-10-24:
I think the best way to implement this may be a parameter for crash detection to be disabled until the motor starts up then if you need to shake it to turn off we should just be able to set another parameter similar to crash detection that's more sensitive.

magicrub, 2016-10-24:
We would need a new param for shake arm/throttleUp, and presumably allows shake disarm, then the crash detector would know about that and react accordingly

jcooper21, 2016-10-24:
Yes that would work much. Is that going to be an easy feature to add?

gmorph, 2016-10-24:
For my launch process there isn't a safety concern.  The plane can't turn on until I active AUTO from from transmitter (or ARM or press the safety button depending on how you have it setup).  However, if your operating without a transmitter and you need to have the plane fully ready to go and then walk somewhere then yes that is a safety issue.
Thanks, Grant.

jcooper21, 2016-10-25:
I think Gmorph might have solved this. I think what he is saying is correct and it might just need to be updated in the APM documentation. The crash detection may work for the Shake off if we have it set correctly. I'll test the Shake off tomorrow and let you know if it will work and if so it will just be a APM documentation update.

jcooper21, 2016-10-25:
So I tested this and I think the shake on will work with that TKOFF_THR_MINACC set to 4.  That should just be a documentation update.  For shake off I think we need another parameter, something similar to crash detection that we can set to the same m/s/s value as shake on.  This could be on a timer so up to 15 seconds after the motor spins up you can shake it off then the crash detection would go back to its normal value.

jcooper21, 2016-10-25:
My crash detection is set to 30 m/s/s, so it will shake off but it just requires more force.

jcooper21, 2016-10-25:
I just did some more testing and I was only able to get the motor to spin up once by shaking. Not sure what's going on.

magicrub, 2016-10-25:
It will only engage if the aircraft is pointed forward. How were you
holding it?

jcooper21, 2016-10-25:
I was pointing it forward fit take off. My flight controller is yawed 270 degrees but it's setup correctly the parameters.

demondev, 2018-03-22:
Is there any progress with this issue ? Anything I can do to assist, coding aside ?

peterbarker, 2025-05-11:
So perhaps we add a flight options bit that allows shake-off.  Shake-off would require `TAKEOFF_ACCEL_CNT`>2.  If the option is set, and we have unsuppressed the motor because of takeoff-accel-detection *and* we're within 10 seconds of unsuppressing *and* we don't think we're flying then quiesce  the motors again.

Does that work logic-wise?

LupusTheCanine, 2025-05-13:
1. IMHO it should be more than 10s, preferably configurable
2. I would check GPS speed < AIRSPEED_MIN or something like that, we need "is_definitely_flying" not "is_likely_flying" as user may move the plane between shaking and realizing they have to abort
3. TAKEOFF_ACCEL_CNT=1 is handy on larger planes as they are hard to shake without increasing sensitivity too much.
4. I would use vertical or lateral (in plane frame) shake to abort launch, a bit like trying to shake something off a tip of a stick. Min accel count for disarm should be 2 (you may get one vertical event during bad throw).