ArduPilot/ardupilot #2222: Plane: smarter WP turn decision to favor into-wind

magicrub, 2015-05-01:
When in AUTO, and you reached a waypoint, and the next waypoint is behind you at least by a certain distance, then the left/right turn direction should take in to account the wind direction to favor into-the-wind. This would reduce the deviation from the planned mission path and favor a sharp turn that results in higher-airspeed rather than a lower-airspeed.

magicrub, 2015-05-01:
@haogodent2 , please post your image here if you have an idea. I can host the image if needed.

magicrub, 2015-05-01:
Here's an image of the intended logic:
http://tompittenger.diskstation.me/~public/turn%20into%20wind%20graphics.png

wp1 and wp2 and ahead so turn towards them liek usual. wp3 and wp4 are behind us and beyond a certain (undetermined) angle and distance threshold where it's more efficient to turn into the wind, left in this case.

Imagine a high-wind scenario where turning right to wp4 sends you way off course and it takes 100+m forward distance is need before you can get back on course which is horrible for a constant-u-turn type mission such as mapping. Turning left will essentially spin you in place and you'd be back on coarse almost immediately.

iskess, 2015-05-02:
I've had a similar issue when mapping at very high resolutions, using very tight flight lines, without decreasing the WP Radius. The WP Radius' of the two WPs that define the turn overlap, and the second one activates before the plane accomplishes a significant heading change. Now the next WP is at the other end of the survey area, and with a crab, the plane can turn the wrong direction, upwind. Now the flight line is ruined.

magicrub, 2015-06-10:
I've narrowed down the code needed to fix this but can't quite get the logic right. I'm open to suggestions and/or help on this.

magicrub, 2015-07-12:
I'm finally getting around to starting on this. Any comments or input?

iskess, 2015-07-13:
Thanks for tackling this. Please test using tight flight lines and large WP radii with a strong cross wind I either direction.

WickedShell, 2015-07-13:
Off the cuff: With regards to WP4 (the worst case and the one that a fix is most interesting for). Compute the lateral distance from the current position to the waypoint, aligned with the axis the wind is blowing on. If the wind speed can push the required distance or more for a given turn performance then turn into the wind, otherwise if the ground speed component + wind component is less then the distance then turn into it. I imagine the difficult part of this is solving the lateral distance a given plane would turn (given it's maximum bank angle). Although since it's a rate controller for the turn this might not be to hard to quantify...

The other advantage is that any solution you find here basically solves the really old bug about having the plane waffle about taking a left or right turn (I've watched it fly about 50 meters before committing to a turn once). (There may still be an edge case but this should be able to come close to solving it).

When you have something to test fly I'd be happy to try it out, we usually fly fairly high winds out here (10 knots gusting 18 is common, and we usually will go to around 15 gusting 20).

magicrub, 2015-07-13:
I thanks for the input guys.

@WickedShell I fixed the turn direction indecision thing a few months ago,
are you still seeing it? I was planning on making the distance based on
your L1nav settings. Ready to assist and a practical distance too.

Technically my intern will be working on this, I'll be advising him. Please
feel free to post any specific SITL settings you'd like him to test
including mission and wind params. More simulations the better for us to
prove it is a rock solid solution.

WickedShell, 2015-07-13:
@magicrub No, I haven't seen it within the last ~6 months (at least not anything that stood out). I missed seeing the commit (and thought the bug was open). Glad to hear it shouldn't be a possibility anymore.

L1 distance... It definitely drives the turn and provides the acceleration demands, not sure if there's a strightforward correlation between that and distance traveled in a turn. (I suspect there is, but I haven't looked at the L1 paper in a while, it might be possible to generalize it fairly easily by assuming a max bank turn).

From some more thought: You can take the max navigation bank rate, and calculate how long it would take to do the turn to the desired angle at that bank rate. Since you now have the time component, you should be able to calculate how far the wind will push you (I believe this can be found with a fairly easy definite integral, or there may be a nice math shortcut). It's also easy to calculate how far in a no wind scenario the aircraft would travel. By combining theses you should be able to figure out if you would overshoot or under shoot/minimize the position error to the waypoint.

I guess that's excessively detailed in some ways, and I haven't thought about it in to much detail, but maybe that helps? (I'd be interested in seeing a simpler solution as well)

magicrub, 2015-07-13:
Take a peek at the L1 code, specifically _L1_dist.

maviday, 2015-07-13:
I'm the intern who's going to be working on this. Hi...

magicrub, 2015-07-13:
Welcome @maviday ! You'll soon be an ArduPilot master!

haogodent2, 2015-07-13:
Instead of current heading as marked by the green line above using the ground bearing of the aircraft would probably be better since the craft will crow.

lekston, 2015-07-14:
Hi all,

@haogodent2 indeed in strong wind the drift angle (crab angle) might be quite large (I've seen more than 45deg drift angle causing "Bad Compass" on multiple occasions even though it the indications were actually correct). On the other hand, when considering turning into the wind, using the current heading as a starting point seems to be more robust. Firstly, when flying with a strong cross wind your the aircraft's nose is already pointing slightly into the wind so there is less turning to be done and secondly, consider the picture provided by magic rub:
  - in conditions like this the ground course of this aircraft might be already pointing to the WP2, so the WP4 would seem to be only 90deg of to the right and making correct decision about the direction of the turn leading to WP4 would be less likely,
 - in other words: turning left to WP4 will require about 180 deg of heading change to the left which will in turn produce about 270deg of ground course change to the left - those numbers describe exactly the same manoeuvre, but it seems to me that the program logic would be simpler when using the current heading as reference.

magicrub, 2015-08-19:
I got something working well but still pounding on it in SITL. More testing is required so I'd appreciate some external participation. This is a critical flight behavior that we don't want to screw up so more corner-case testing the better!

Branch:
https://github.com/Airphrame/ardupilot/tree/uTurnIntoWind

new param: NAVL1_ANGLE_THR
// @Units: degrees
// @Range: 90 - 179, recommend 120

Jman841, 2016-11-06:
Any updates on this? I had a windy day today and when the plane turned with the wind, it ended up wayyyy off course. This could have been easily avoided if it turned against the wind instead of with it.

magicrub, 2016-11-07:
no progress on my end, I've been doing other work. I'd really like this feature though.

Naterater, 2019-01-04:
I think this is still very relevant. It can cause very significant deviations from waypoint paths during course-reversals. 

To begin, the simple logic would be to estimate the new **heading** to make it to the next waypoint. The estimated wind speed must be factored in . Then take the shortest turn based on the current heading. Right now it's based on bearings, and that is not correct.
[image]

JeyPi, 2023-01-13:
Hi. I confirm the turning in to the wind is something I would like very much. I do not understand why the Arduplane decides to turn with the wind when it is already flying a bearing that is better to turn in to the wind. See my picture. Numbers are waypoints in sequence. Sometimes it turns in to the wind sometimes it does not.
[image]
[image]

peterbarker, 2025-05-11:
I've taken WIP off this - pretty sure it's not "In Progress" ;-)

I've also removed the 3.9.0 milestone - that one was missed a touch.

I've also added the "Task" label as I think the problem is well-enough spelt out in this issue that someone could complete the task.  I'm also quite sure convincing the L1 controller to do this will be challenging.