ArduPilot/ardupilot #2245: enhancement: log rssi to TF/DataFlash

KIrill-ka, 2015-05-06:
Please, consider logging of receiver RSSI. Possibly in the RCIN message...

magicrub, 2015-05-06:
Hi Klrill-ka,

RSSI (and others Rx data) is stored under "RAD" but currently it is only logged when a MAVLINK_MSG_ID_RADIO or MAVLINK_MSG_ID_RADIO_STATUS mavlink message is received and the MASK_LOG_PM bitmask is enabled (bit 0x80). Source code is here for plane but all platforms have it.

There must be a good reason why it was implemented to only log on a certain mavlink packet instead of periodically, does anyone know why?

magicrub, 2015-05-06:
Hi Klrill-ka,

RSSI (and others Rx data) is stored under "RAD" but currently it is only
logged when a MAVLINK_MSG_ID_RADIO or MAVLINK_MSG_ID_RADIO_STATUS mavlink
message is received and the MASK_LOG_PM bitmask is enabled (bit 0x80). Source
code is here
<https://github.com/diydrones/ardupilot/blob/master/ArduPlane/GCS_Mavlink.pde#L1570>
for
plane but all platforms have it.

There must be a good reason why it was implemented to only log on a certain
mavlink packet instead of periodically, does anyone know why?

-TomP

KIrill-ka, 2015-05-06:
The RAD message is for telemetry modem, isn't it? I mean RSSI for RC receiver.

magicrub, 2015-05-06:
ahh, you're right. So yeah, we should probably add this somewhere.

magicrub, 2015-09-16:
Done via https://github.com/diydrones/ardupilot/pull/2804

closed 2015-09-16 by magicrub