ArduPilot/ardupilot #2247: Arduplane: Auto Tune Yaw, L1 nav, etc.

Jman841, 2015-05-06:
Hello,

I was wondering what the current progress is on the development and refinement of the Auto Tune function and if auto tune will soon be added for Yaw, L1 nav, and other important control loops that are currently manually tuned?

Thank you

iskess, 2015-05-13:
+1 I find yaw control the most difficult to tune. There are lots of reports of planes flying in an uncoordinated attitude, which results in an offset from the planned flight line. The carbon boomed Skywalker is particularly prone to this, and since 3DR sells the Aero, it would be very productive to put some resources towards a customer friendly solution.

magicrub, 2015-05-13:
+1
Current auto tune is useless for my Skywalker x8.

EShamaev, 2018-11-11:
@magicrub How about tune mode when plane is put in a low radius loiter and tune PTCH2SRV_RLL with error from pitch and by averaging error on Z/Y acceleration - tune YAW2SRV_SLIP and/or YAW2SRV_INT.
Open question - what to do with KFF_RDDRMIX and YAW2SRV_DAMP?

magicrub, 2018-11-12:
I like the idea of a loiter doing an autotune.. more automated. However, a real tune really needs some aggressive inputs to compare against. That could be mixed in. Another idea is to implement a built-in hardcoded mini-mission of a racetrack. Loiter radius on each end and then straight for a few times the radius distance.

I'll ping @tridge about doing self adjustments of those params

magicrub, 2018-11-13:
result from devCall meeting: easy to ask the question, hard to get a solid solution. Pull requests welcome!