ArduPilot/ardupilot #2251: wishlist: Allow setting mode switch to -125 (past RTL) to cause loud noise in speaker for lost aircraft

marcmerlin, 2015-05-07:
Currently the mode switch, if it goes past -100, isn't recognized as RTL with standard setups.
It would be great to allow recognizing  a 7th mode, past RTL, that would simply cause the beeper to get very loud to find the model after a crash in vegetation.
I'm currently using a 3rd party model finder plugged into the same port, and have that functionality, but it would be great not to have to depend on yet another device when the pixhawk is totally capable of outputting loud obnoxious sounds if asked :)

magicrub, 2015-05-07:
This has been implemented in AC3.3 by setting an aux switch to AUXSW_LOST_COPTER_SOUND

marcmerlin, 2015-05-10:
Awesome, thanks @magicrub .  I see it's not documented yet, but now that you gave me the name, I can find hits in the code.
But is it just in copter? I have a plane and it could get lost too :)

magicrub, 2015-05-10:
I wrote one for plane, which inspired the one for copter, and I think
there's still a pull request pending but I ended up having some problems
with it and lost interest so I may have closed it.

marcmerlin, 2015-05-10:
Is it fair to leave this open until the feature is both in plane and copter then?

rmackay9, 2015-05-11:
I think we already have two methods of triggering the lost copter alarm.  A ch7/ch8 switch and a special stick movement (i think it's left stick down-left, right stick down) so I don't think we will implement a 3rd way to trigger the alarm.

marcmerlin, 2015-05-11:
@rmackay9 a 3rd way is not needed, I wasn't aware of the current code when I made my suggestion. Just hoping for this to also end up in the plane source tree :)

marcmerlin, 2015-05-26:
@magicrub: do you mind re-opening this bug since it's not done in the Plane tree yet?

magicrub, 2015-05-26:
The issue with my pull request was I was trying to get fancy by having the
alert turn off automatically when you picked the plane up. I can strip out
that stuff and reopen.

marcmerlin, 2015-12-06:
@magicrub 
Do you mind re-opening this issue as a reminder to add the functionality in plane too? Thanks :)

magicrub, 2015-12-06:
Alarm after land until nice change seems like a reasonably simple feature.
Would that work?

iskess, 2015-12-06:
I presume you meant "mode change"?

Personally I would like something that activates the beeper for unplanned landings or crashes. After a crash, the first thing I would do is switch to Manual to prevent the motor and servos from moving. If this deactivates the beeper, that would not be helpful. 
I like the -120% on the mode channel idea. Or perhaps if you hold the rudder/throttle in the disarm position for 10 secs the beeper activates.

magicrub, 2015-12-06:
Lol, ya. Sorry, auto correct killed me there.

That's true, switching out of auto right after a land is common procedure.
You only really need the beep if you don't know where it landed so a
commanded alarm makes better sense. So, I like the alarm via long disarm
but want to make sure there's a way to turn it off. Perhaps mode change?

iskess, 2015-12-07:
I would turn it off by using doing the opposite stick position. By putting the rudder in the arming position. That's the most intuitive solution. I don't see a big risk in arming the plane once you've found it.  Usually you would just unplug the battery once you find the plane.

rmackay9, 2015-12-07:
In copter, the alarm only sounds for 1 or 2 seconds so the pilot needs to either keep the TX sticks in that special position (both down and right) or keep the ch7/ch8 switch in the on position.

magicrub, 2015-12-07:
Yeah, I like that even better! Just hold rc input as disarm and it beeps.
Release and it ends. Easy and safe and intuitive.

iskess, 2015-12-07:
I'd rather it stay on without holding the stick, but I'll just make a mix in the Taranis and link it to a switch and I'll be happy :)

marcmerlin, 2016-03-29:
Look at this again, Plane now allows disarming with long rudder to the left (like copter). 
It seems that if you just kept the long rudder left longer, you could start an alarm.
Another option is you could just sound the alarm after X seconds if the plane has determined it's not flying anymore, but  it's still armed (slightly better because it would work if the RC receiver has been ripped out in a crash, but ardupilot is still powered on).

In the meantime, if others care to get this, I'm using 3 other ways (yes, I'm that adverse to looking for my plane in tall weeds ;) ):
- I bought a self contained keyringer https://www.keyringer.com/ and taped it to my plane. I can trigger the sound from my cell phone or another keyringer. This works great even if all electronics have been unpowered/battery unplugged
- I now use a brotronics openlrs module with backup lipo and buzzer on the receiver (all taped together). So this works separately from main power and will buzz on signal loss as well as send rescue me audio beacons on 433Mhz that you can receive with an FRS radio
- I have a lost model finder buzzer plugged into an RC channel. I can enable that one if I set mode switch to -120% (past RTL). This only works if everything is still plugged in though.
- FPV goggles and/or 3DR radio record the last GPS location received before crash

Hope this helps others. Allowing the system buzzer to be used as a locator would still be cool though, just not as important to me anymore.

peterbarker, 2025-05-11:
So ArduPlane definitely supports a "Lost Plane Sound" you can attach to an auxiliary switch.

I guess this could come down to having an option bit set.  If the bit is set and you ever become non-flying then it starts to sound an alarm; disarming or changing modes would silence the alarm.  We should probably also sound the alarm when crashed....

peterbarker, 2026-11-15:
We should reuse some ideas and/or code from #33096 here