ArduPilot/ardupilot #2255: ArduCopter AC 3.3 rc 3 -  Bad Accel and Bad Gyro - Persist on AC 3.2

andrea25, 2015-05-07:
I was testing AC3.3 rc3.
I was testing it with a new PixHawk. Original PixHawk by 3DR.

I have already re-done the Accel calibration and the Compass calibration on AC 3.3 rc 3

Some times appen that I see "BAD_ACCEL_HEALTH" on the planer.
Some times appen that I see "BAD_GYRO_HEALTH" on the planner.

Every time I see the problem I have to unplug the battery, wait some second, and plug in the battery.
After some repetition of this, the problem is solved automaticaly.

The problem persist with AC3.2 
(Yes, when I downgrade I redo the Accel and the Compass calibration).

It do not persist with AC 3.1.5.
I think it is a problem in reading something with IMU2.

It is fixable via software?
The PixHawk is working and it is brand new.

rmackay9, 2015-05-07:
It's hard to be sure without seeing a log but it sounds like a hardware issue and I'd recommend doing an RMA with 3DR.

closed 2015-05-07 by rmackay9

andrea25, 2015-05-07:
It is not this?
http://diydrones.com/forum/topics/pixhawk-bad-accel-health-apm-3-2-collecting-data

Unfortunately I did not have ARMING/DISARMING log enabled when the issues appened.

I have some other log, but in these last log everything was fine.

rmackay9, 2015-05-07:
@andrea25, Are you ThatJoshGuy or is your log is somewhere on that thread?

andrea25, 2015-05-07:
Sorry, it took me a long time to respond because I was browsing in my log.
I browsed a lot in the logs that I made in recent days, and I had found a log that I did indoors.

This is the problem that I have written in this issues.

I disabled the prearming check.
This test was done indoors.
I did this test to log the problem

I was trying to fly in "stabilize". I raised my hexacopter of 3-5 centimeters from the ground, and after I landed.

The IMU 1 is not flat, the IMU 2 is flat.
[image]
[image]

How to attach the full log? Sorry, I'm a newbye in GitHub. I can not do it correctly.

I'm not able to reproduce this issue.
Sometimes it comes, sometimes not.

--------Reply to your question----
No I'm not ThatJoshGuy.
And I've never posted in that thread.
I'have discoverd that thread by looking some solutions for my issues.

Thank you for everything (=

andrea25, 2015-05-07:
Here there are the full log (=
http://www.mediafire.com/download/w14sukewyrdevxd/2015-05-01_22-10-20.log

rmackay9, 2015-05-07:
@andrea25,
Ok, so that's a failure of the lsm303d.  It's not the issue that 3DR manufacturing was having from June-2014 ~ Feb-2015 that affected the MPU6k.  Still, it's a hardware failure and if it's a relatively new board from 3DR I would expect they would do an RMA to give you a replacement.

andrea25, 2015-05-07:
Thank you very much for yout kind reply.
I have understand.

Now I contact the customer care/support center to see what I can do about this hardware problem.

Thank you very much!