ArduPilot/ardupilot #2266: AHRS_TRIM not set correctly by accelerometer calibration for some non-zero AHRS_ORIENTATION

brad112358, 2015-05-09:
For reasons of wire routing, my Pixhawk is oriented with a 180 degree yaw.  Thus, I have AHRS_ORIENTATION set to 4.  After I perform accelerometer calibration on a carefully leveled surface, telemetry always shows approximately 3 degree of roll when on the same level surface.  This is reproducible for every calibration and, of course, causes constant drift in stabilize modes.  I can work around this issue by setting AHRS_TRIM_X and AHRS_TRIM_Y to values close to 0 by hand.

I experimented with setting AHRS_ORIENTATION to 0 and performed another calibration, now treating the back of my craft as the front.   This resulted in very level readings (within .04 degrees) on the reference surface.

It looks like something might be wrong with how orientation is handled in the calibration code.  I have not tested other orientations.  These tests were performed with AC3.3rc3, but I am not aware of changes to this area of the code since the last release.

brad112358, 2015-05-09:
It looks like the problem is near the end of AP_InertialSensor::calibrate_accel().  The rotation should be done *after* applying the calculated scaling and offsets, not before as the current code is doing.

I will work up and test a fix and issue a pull request.

prickle, 2015-05-10:
I am also having this issue.
I am using AUAV-X2 mounted upside down, requiring AHRS_ORIENTATION=8. After calibration the horizon leans like a drunken sailor and I get Bad AHRS messages. Reverting to 3.2.1 restores normal behavior.

brad112358, 2015-05-10:
prickle,  I've recently seen the behavior you are describing as well, but I don't believe it is related to the problem I reported here.   This issue only causes small constant errors in roll or pitch.  You may want to file another bug.

prickle, 2015-05-10:
brad112358, you may well be right. However, I feel they are related. The unstable horizon happens after calibration and before rebooting. After rebooting the board it seems to become a steady RP offset from level as you describe.

prickle, 2015-05-12:
Sure thing brad, I'll give it a go. I have not much free time though so you may be waiting a few days for me to get it together enough to get results. Mind you, I'm guessing the developers may find this one hard to ignore and easy to fix. Depending on their workload I would not be surprised to see it brought in in the next few days. Either way I will be compiling it sooner or later to try against my issue.

prickle, 2015-05-13:
So now I have built brad's fix and can confirm it works for me. The horizon is now level and correct after doing a calibration followed by a reboot. Thanks brad :+1:

brad112358, 2015-05-18:
Fixed by #2295 and also b564ba0 in ac3.3rc4
Thanks!

closed 2015-05-18 by brad112358