ArduPilot/ardupilot #2268: to low limits of rate roll and pitch values

buckker, 2015-05-09:
I have the problem that the limits of rate roll and pitch values are too low. I fly a X8 Copter with the following values:
rate roll and pitch P: 0.39
rate roll and pitch I: 0.6
rate roll and pitch D: 0.01

In the version 3.2 I can overwrite the values manually in the full parameter list. Under 3.3 RC2 it's not possible (or I do something wrong)

What's the goal of those limits? For me it's a problem because the copter fly with lower values not stable enough. Is there a possibility to disable those limits?

many thanks in advance

Michael

R-Lefebvre, 2015-05-11:
Are we somehow limiting these parameters in Arducopter?  I would be surprised if that were the case.

I wonder if this is more of a Mission Planner issue?

magicrub, 2015-05-11:
I heard @rmackay9 mention once that some values were limited but I have no idea if that applies to these specific values or not

buckker, 2015-05-12:
Hi

I think you're right. The limits comes from the mission planner. For expample, for the rate_pit_p the pre defined value is from 0.08 up to 0.3. You can't go higher than 0.3 in the extended tuning.

Should I open a new issue in the mission planner part on Github?

R-Lefebvre, 2015-05-12:
That's definitely an MP issue, and you should open an issue on Github.

But I really don't know what the solution is. Those protections are there to keep the majority of people out of trouble.  So it's hard to figure out how we can protect the many without restricting the few.  Maybe we need a check-box to obey those suggestions or not.

magicrub, 2015-05-12:
Can you enter any value in the full list?

R-Lefebvre, 2015-05-12:
You should be able to.  If not, that's a problem.

Again, this gets back to the decision between protecting some people at the expense of making life difficult for others.

buckker, 2015-05-12:
I can understand the goal of those limits. For pilots without big knowledge it's a good thing for sure. But for the others it's a big disadvantage. I'm working with ardupilot because it allows me to modify a lot of things according to my experience.

Probably two mission planner modes could be the solution. One for the "beginner" and one for the specialists.

In the full parameter list I can enter higher values. But in the extended tuning it's no possible.

R-Lefebvre, 2015-05-12:
Yes, it would be nice if the restriction could be relaxed in the extended tuning page somehow.  I also have to work with the full list most of the time, and it's less than ideal.

buckker, 2015-05-12:
Good idea :-) In the full parameter list are to many values. Its hard to find the correct value to modify at the first glance..

rmackay9, 2015-05-13:
So we have two choices:
1. increase the acceptable range in the parameter descriptions
2. ask MichaelO to make i easier to go beyond the acceptable range

If everyone agrees we can increase the acceptable range without putting more people in danger I can do that.  So it's a judgement call.. will more people get themselves into trouble by making the gains too high or will more people be annoyed by not being able to set the gains high enough?

rmackay9, 2015-08-22:
I'm going to close this issue because we haven't had any further feedback and I think the parameters can be updated in the mission planner's full parameter list.

closed 2015-08-22 by rmackay9