ArduPilot/ardupilot #2270: ArduPlane: Support for Digital servos

Jman841, 2015-05-10:
Could there be support for higher frequency refresh rates for digital servos? 333hz instead of only the default 50hz?

We are having issues with some digital servos in one of our planes where it will bind itself on its way back to center. It's not a mechanical bind as the servo is completely free when testing. We are almost positive it is an issue with the 50hz refresh rate being used on a servo designed for 333.

R-Lefebvre, 2015-05-11:
Interesting issue.  I'd be surprised if the refresh rate was the issue, but I'm interested to find out.  Arducopter Tradheli use 125Hz by default, but I would like to allow 333Hz.  

One issue, I believe with Pixhawk, is that you have a limited frequency setting available.  ie: some channels are tied together.  1-4 5-8 and 9-12.  Something like that.  So if you go to high frequency for 1 servo, you will also give it to other channels.

Jman841, 2015-05-11:
It may be a different issue, but if it is we can not figure out what is causing it as it is present in the two aileron servos and the elevator all using the same servo type. However, with our rudder we are using a different digital servo and have no problems at all.

It seemed to be a similar issue as if you set the tail servo on a Heli to the wrong frequency which is what lead us to believe it was a refresh rate issue.

I think it would be ok to have the option of setting the refresh rate the same for each series that you mention. With the plethora of low cost, high quality digital servos now available, I don't see any reason why it would not be a benefit to support their supported refresh rates.

R-Lefebvre, 2015-05-12:
I wonder what percentage of digital servos can accept 333Hz?

I don't actually believe 333Hz vs. 125Hz (or even 50) makes much of a noticable difference in flight quality.  I think it was more of a marketing thing for the servo manufacturers.  So I don't think this is a functional improvement.  But, if modern digital servos are starting to not operate on 50Hz, then we have a problem that needs to be solved for sure.

I'm currently operating a very high speed, low-bandwidth (760uS) heli tail servo on Arducopter at 125Hz no problem.

Can you use 125Hz on Arduplane?

Jman841, 2015-05-12:
Unfortunately, on Arduplane only 50hz is supported. I agree that 50hz is more than enough, performance wise, for a plane but also have the same concerns about the servo not interacting well with this frequency. Maybe offering the option of supporting different frame rates? 50hz, 125, 250, 333? 

Does this increase processing load substantially or is it difficult code to write? I know Arducopter for multirotors currently supports 400hz so I am assuming it's not too demanding on the processor. 

I would love to take a stab at writing a pull request myself, but, don't currently posses such skills

R-Lefebvre, 2015-05-12:
Yeah, I think this isn't a problem for the processor, and is probably a fairly easy change to support 125Hz for plane, since this is all done in the HAL layer which is platform-agnostic. And we know it already works on copter. It might be just a couple lines of code.

Where I'm less certain, is how hard it would be to do 333Hz. There might some reason why only 50, 125 and 400Hz are possible.  Not sure.  I'd like to look at it for copter, but unlikely I would make a change to plane.

magicrub, 2015-05-12:
I believe 400 is used because a lot of processing happens then so the pwm
is updated then too. Plane does not need beyond 50. Anything above that is
just marketing. However, since the popularity of mini quads, receivers may
be generating higher frequencies so at least being able to support them may
be worth while from, again, a marketing and adoption point of view.

Jman841, 2015-05-12:
I agree, however, the concern is if some digital servos are not compatible at a 50hz refresh rate which may be the issue we are seeing with our current setup. It seems there would not be any negatives to supporting higher refresh rates on planes in case this happens for a specific set of servos. Performance wise I agree, for a plane application there would probably be no noticeable improvements

R-Lefebvre, 2015-05-12:
This is the line that ultimately sets the refresh rate on helicopters:

hal.rcout->set_freq(mask, _speed_hz);

I'd look for that in the plane code, maybe change it, and then compile.  See if that works.  Just as a test.

Jman841, 2015-05-12:
would this be the parameter I would have to change? and I apologize but I am not sure how to compile a test firmware.

https://github.com/diydrones/ardupilot/blob/42c6801d61c653ec4958d3eb68b56307b19c9a1f/libraries/AP_HAL_SITL/RCOutput.h

If so, would it be possible to add this parameter within Mission Planner for the user to be able to input. Keep the default as 50hz, but the user can simply type the refresh rate of their servos if experiencing a problem with the 50hz refresh rate?

R-Lefebvre, 2015-05-12:
No, sorry, that's not it.

We probably need to get Tridge or Tom on this.  It would probably be easier to get their attention if we were sure this is the problem.

I wonder if you could try loading helicopter code on the board temporarily, and set H_Servo_Man to 1, put it in manual mode.  Then see if the servos move properly at 125Hz.

Jman841, 2015-05-12:
Ok,

We are switching the servos out today. The ones that are giving us problems are Gaui branded GS-311 Digital servos. I believe they are Savox rebrands.

Were going to throw some MKS DS-95's in to see if they work better as we have a Hitec 5055MG on the rudder and that servo has no issues.

Once we get them out I will test with a spare pixhawk on the traditional heli firmware and see if I can replicate the problem.

Thanks for the help!

R-Lefebvre, 2015-05-12:
For sure, that's seems like a great approach.  I'm sure if you can prove there's a compatibility problem with the servos on 50Hz, Tridge will help you out.  This should actually be a pretty simple change, just adding a parameter for servo speed and a setter function on init.

Jman841, 2015-05-12:
Switched to the DS95's and now we have no issue. Something with those other servos were not happy with the refresh rate.

GitLang, 2015-07-17:
"I'm currently operating a very high speed, low-pulsewidth (760uS) heli tail servo on Arducopter at 125Hz no problem."

 R-Lefebvre  -  could you give any more details on this please?. Is the featuree available in the Beta, if so how to enable it please?. Thanks.

GitLang, 2015-07-24:
Rob, you appear to have missed my question.  I am only asking you to share some knowledge, or thew source of that knowledge. Thanks.

"I'm currently operating a very high speed, low-pulsewidth (760uS) heli tail servo on Arducopter at 125Hz no problem."

R-Lefebvre - could you give any more details on this please?. Is the featuree available in the Beta, if so how to enable it please?. Thanks.

magicrub, 2016-06-06:
is this still valid?

Jman841, 2016-06-07:
I believe so, As far as I know you can not change the output frequency for the servos in Arduplane still. I don't see any disadvantage to supporting a higher refresh rate as digital servos seem to be the norm now. I would use it if supported as the servos we use support up to a 333hz refresh rate.

R-Lefebvre, 2016-06-07:
Sorry GitLang, I missed your question.

Using 760uS servos is only possible on helicopters, because it is the only vehicle type that has some degree of separation between the RC Input and the RC Output channels.

To use a high speed tail servo on a heli, simply set HS4_MIN and HS4_MAX to 500 and 1000 (or whatever suitable endpoints you need).  This moves the pulsewidth into the range needed. Done.

GitLang, 2016-06-07:
Many thanks for the reply Rob, that is indeed good news. The project has been on the back burner while sorting a different problem and I am due to come back to it next month, so the timing is perfect!

magicrub, 2016-06-07:
I know we support One-Shot at higher rates but I don't see it in the wiki yet so I don't know where to point you to configure and test it. Not sure if we can de-couple the higher-rate with the smaller pulse of one-shot. Would be nice, does anyone have any insight on this?

lvale, 2016-06-07:
I believe Tridge enabled higher frequency only on the SBus out. Enabling higher frequency for the other pins can be a risky issue, I 've had my share of burnt 50Hz servos because I plugged' em on a plain Flybarless system that was setup to 333Hz...

R-Lefebvre, 2016-06-07:
On Arducopter, we can set the servo rate. It's just a parameter, been there a long long time.  Why is it not available in plane?  I thought it was in the HAL or generic layers, so available for all vehicles?

magicrub, 2016-06-07:
what parameter?

R-Lefebvre, 2016-06-07:
RC_SPEED

Naterater, 2018-12-28:
This is available via SERVO_RATE: Servo default output rate.

closed 2018-12-28 by Naterater