ArduPilot/ardupilot #2273: AC3.3rc3 AutoTune Yaw can save out of recommended settings

bionicbone, 2015-05-10:
Hi,

I tried AutoTune on a x8 frame, shortly after completion I changed the battery then crashed. 

I'm not convinced that it was due to the AutoTune, its a toss up between a broken prop 5 or what I have discovered below...

After rebuild I came to enter the last AutoTune settings and noticed Yaw RATES and STB values were higher than recommended settings for the YAW axis.

APM Log Analyser v2.1.0.8 results (I have validated this is Mission Planner)
AutoTune Recommended Yaw Settings:
   Stabilize Yaw = 6.648549
ERROR: The above data is out of recommended range!
   Rate Yaw P = 0.7585834
ERROR: The above data is out of recommended range!
   Rate Yaw I = 0.0758583411303774
ERROR: The above data is out of recommended range!
   Rate Yaw D = 0
   v3.3-rc3: Yaw Rate D is not saved (found 7.22)
04/05/2015 09:31:49 - 243973: AUTOTUNE_SUCCESS
04/05/2015 09:32:11 - 249944: LAND_COMPLETE_MAYBE
04/05/2015 09:32:12 - 250156: Landed

76.bin is the AutoTune Log
77.bin is the Crash Flight Log
https://www.dropbox.com/s/6iljmadyo6f1jy1/76.BIN?dl=0
https://www.dropbox.com/s/eiyb7nhnmvp6chv/77.BIN?dl=0

When you compare the Params between 76 and 77 we can see that RATE_YAW_FILT_HZ was also changed from 5 to 7.22 -  does AutoTune alter this?

The crash in 77.bin looked like I broke Prop 5 (as this part was not found at the crash scene), hence this could be teh actual crash reason.

However, curiosity got the better of me so after the re-build I entered the AutoTune settings again to see what these settings would do. I notice that all seems well until the user enters some yaw, "only a little" in stabilize mode causes the X8 to gain altitude aggressively (although she stays level'ish). The motors sounded like they are pulsing aggressively also. I did not dare to increase yaw at this stage.
87.bin shows this flight.
https://www.dropbox.com/s/acgjtprdze8au6j/87.BIN?dl=0

I'd be pleased to hear your thoughts.
Kev

lthall, 2015-05-10:
This isn't where you should be asking these questions. 
However you have not used the voltage scaling so your yaw tune may be a bit hot. Looking at your Yaw tune, everything looks good. I would suggest you reset your yaw parameters and do the yaw tune on a full battery.
This should not be able to crash your copter though. So I suspect your are correct that you lost a prop.

bionicbone, 2015-05-10:
Hi Ithall,

Sorry, its always difficult to know where to post and I thought this is an issue worth reporting.

Bottom line, v3.3 AutoTune did not do so well determining the Yaw settings and I am still convinced that if I pushed harder on this axis with these settings the results would not be good.

The questions I ask were less important to me than giving you the information of what I have observed with a release candidate. 

Thanks for looking, I'll set AutoTune to only tune Yaw on a full battery and try again.

Thanks

lthall, 2015-05-10:
Hi bionicbone,
At the moment the 3.3 thread is the way to go for problems like this. GitHub is for known bug reporting. And don't misinterpret short replies as unfriendly, we do this a lot and tend to get to the point quickly.

closed 2015-05-11 by rmackay9