ArduPilot/ardupilot #2274: px4flow i2c problem with OPTICAL_FLOW_RAD messages unhandled.

deb0ch, 2015-05-10:
Hi guys,

We've been trying to use the px4flow on our pixhawk running arducopter, but our major problem was that the i2c bus was so polluted that we suffered from lots of data losses and delays: sometimes the signal got blocked for seconds and even tens of seconds !

We found out that there was that OPTICAL_FLOW_RAD mavlink message, which seemed to us like a perfect clone of the OPTICAL_FLOW message, being sent at the same time as the OPTICAL_FLOW messages to the pixhawk, and only one of the two were consummed on the ardupilot (arducopter) end.

The fix that we found and spectacularly resolved all of our problems was just commenting the lines where that "clone" message was sent over i2c.

I don't know what is the primary use of that message, if it is a suitable way to fix the problem or if that problem should be addressed on the ardupilot's end or on the px4flow's end, so that's why I'm signaling and asking you.

I posted this issue on the px4flow's repo as well because as I said, I do not know on which end the problem should be addressed

Cheers.

Naterater, 2019-01-07:
Any updates @deb0ch?

deb0ch, 2019-01-07:
Absolutely not 😅 We stopped using the px4flow years ago ant went on to other projects, so I don't think there ever will be any updates on my end.

With this issue I was just signalling a problem and a way to fix it, as a civil duty, in case anyone would be interested.

rmackay9, 2019-01-09:
This issue seems odd though because MAVLink messages are sent over serial not over i2c.

closed 2019-01-09 by rmackay9