ArduPilot/ardupilot #2277: Tarot Gimbal Tilt not working on ArduCopter 3.3 RC3

zumbrunnen, 2015-05-11:
I've upgraded to the last RC version and the Tarot Gimbal's tilt (configured as shown on http://copter.ardupilot.com/wiki/common-optional-hardware/common-cameras-and-gimbals/common-tarot-gimbal/) does not work any longer. The blue led stays solid, instead of flashing blue indicating it's connected to Pixhawk.

Reverting back to 3.2.1 solved the problem: flashing blue led and camera tilt is controllable again.

Comparing the current configs with backups and default params for the frame (X8+), I've noticed an absent `MNT_MODE` (was at a value of 3) in older backups (>= 3.2.1). Did this get changed to `MNT_TYPE` in ArduCopter 3.3 RC3? Anyway, I played around with `MNT_TYPE` but the gimbal's tilt control did not come back to life.

rmackay9, 2015-05-11:
@zumbrunnen, can you try setting the MOUNT_TYPE parameter to "1" and reboot the board?
A couple of other people have bumped into this and it's mentioned in the top part of the  AC3.3 beta testing thread.. but there are a lot of words there and don't necessarily expect everyone to read them all.
Thanks for testing!  I'll monitor this issue in case you find it doesn't work.

closed 2015-05-11 by rmackay9

zumbrunnen, 2015-05-11:
Thanks @rmackay9 - problem solved!

I'm sorry, didn't check the diydrones.com thread. That forum is not very convenient to follow nor to search for something in threads. At least this "problem" is now documented here :wink: