ArduPilot/ardupilot #2278: add "@ReqReboot: true/false" to all params

magicrub, 2015-05-11:
Due to the varying complexity of params, some work instantly while others require a reboot to work correctly (or at all). To document this, lets add a "@ReqReboot: true/false" field to every param.

rmackay9, 2015-05-11:
Related to this issue: https://github.com/diydrones/ardupilot/issues/2199

A great idea.

magicrub, 2015-05-12:
here is a brief list of ones that need it
MOT_SPIN_ARMED
SERIAL*

chapman, 2015-05-12:
+FRAME

magicrub, 2015-05-12:
Lets do a tri-, or more, state scheme and change the label to @ "TakesEffectAt:" with possible values of "Immediate", "immediate while disarmed", "AtReboot"

hsteinhaus, 2015-05-12:
Some params require a mode change to take effect, e.g. WPNAV_LOIT_SPEED

magicrub, 2015-05-12:
Even while disarmed?

hsteinhaus, 2015-05-12:
No, during flight. This behavior seems to be a thing between "immediate" and "immediate while disarmed". Some Auto-related params show the same behavior.

magicrub, 2015-05-12:
So it takes effect while disarmed? Adding a forth "on mode change" may add
another layer of complexity that we like to avoid. I would vote to not have
that as a choice and to just roll out into disarm. The exception is of this
is something someone would like to change during flight.

R-Lefebvre, 2015-05-12:
I think this issue with Loit Speed is an issue that needs to be fixed so it's immediate, rather than just documented and managed in the params.

hsteinhaus, 2015-05-12:
Probably it will. I was never tempted to try that in the middle of a 1hour Auto mission, e.g. just for changing the WPNAV_SPEED from 8 m/s to 7 m/s after a wind change. It would be absolutely nasty (almsost show-stopping) if the GCS would force me to do more than a mode change here. 

IMHO the complexity is not added by describing the behaviour precisely, but by the behaviour of ArduPilot itself.

magicrub, 2015-05-12:
As Rob said, we're better off changing the param to instant if we can fix
it to be so. Long-term goal is to get all params to do that where possible.

-TomP

hsteinhaus, 2015-05-12:
Ok, that would be fine. But I think there is a whole bunch of parameters that show the same behavior because there is a very common pattern in the code. Those parameters are read one time when the flight mode is initialized and copied to local variables and/or used for initilializing other sub-components. If the param changes thereafter, there is no instant effect. I would not expect there will be an easy fix.

R-Lefebvre, 2015-05-12:
Actually, I think most parameters will update their global variables immediately (probably all do).  The problem is some of the parameters get used to generate other static variables in init functions that only run on boot up.  We have been trying to move many of these init calculations into a periodic process that runs in the 1 second loop while disarmed.  But some are either tricky to do, or we don't want them to take immediate effect for other reasons.

For example, the loiter speed calc affects the leash length, which is only calculated once when initializing the mode.  Perhaps on Pixhawk, we could do this calculation all the time, but we'll have to look into that.

rmackay9, 2015-05-28:
I've added this parameter description to the RCMAP parameters after svefro made it clear we should have this.
https://github.com/diydrones/ardupilot/commit/19b021f72922eda13508058a2895d31e11bdf164

WickedShell, 2015-07-20:
This is needed for most of the GPS parameters, I'll submit the PR after the GNSS one is done.

amilcarlucas, 2017-08-28:
This is better now

magicrub, 2017-08-30:
We've done this in many places. Probably safe to close it for now

closed 2017-08-30 by magicrub