ArduPilot/ardupilot #2279: Arduplane Helical Landing Pattern

Jman841, 2015-05-12:
I am not sure if this is being worked on, but, it would be very nice to have a helical landing pattern that commands the plane to spiral down to the landing altitude in a specific size radius and then level off and finish the landing. I think it may make landing much easier for some fields that are surrounded by mountains or tall tree's and don't have the clearance to do long gradual altitude changes over a large distance. The user could just select the landing type (Helical), the direction of final touch down, the pitch angle or sink rate, and the radius or diameter of the helical pattern (as long as it is within the airframes capabilities).

magicrub, 2015-05-13:
This can be done with a combination of mission items.

Normal way point up high
Loiter until alt (low) and angle match (to land wp)
Then land

magicrub, 2015-10-22:
There is a discussion about this here
https://groups.google.com/forum/#!topic/drones-discuss/Q5_MivrAzBY

Consensus so far is to use LOITER_TO_ALT with mission params heading lock and add a sink rate.

davidbuzz, 2015-12-26:
MissionPlanner also has a "Scripts" interface, and a few examples available, one of which is a script called C:\Program Files (x86)\Mission Planner\Scripts\PARACHUTE_LANDING_APPROACH.py  that I suspect would make an excellent example to modify for the purpose of a helical landing approach.

magicrub, 2016-01-29:
I have some very good progress on reverse-thrust landings and testing helical landings with it

magicrub, 2016-06-06:
Helical turn landings work muuuch better with new param [LAND_SLOPE_RCALC](http://ardupilot.org/plane/docs/parameters.html#land-slope-rcalc-landing-slope-re-calc-threshold)

WickedShell, 2017-05-29:
Retagged as Plane 3.9, with AP_Landing this should be a lot easier to implement.

Naterater, 2019-01-03:
@magicrub, didn't you do an example of this at the last Developer meetup about a year ago?

magicrub, 2019-01-03:
@Naterater Are you referring to this?
https://drive.google.com/file/d/0B6RPB3yIS_0MZEt4aUtkOGpzZFE/view?usp=sharing
and
https://youtu.be/XuSl9Io93aQ?t=1640

magicrub, 2019-01-03:
it's possible now, but I'd like to add this capability as a built-in feature instead of hacking at mission and params to make it happen. This may end up better done as a script

Naterater, 2019-01-17:
Should this stay open until scripting is complete then?

closed 2021-07-26 by Naterater