ArduPilot/ardupilot #2280: Travis: add param description tag check

magicrub, 2015-05-12:
Travis: add param description tag check. This will be helpful to determine if param description and TakesEffectAt items are included. Build should fail if not found

related issue https://github.com/diydrones/ardupilot/issues/2278

magicrub, 2016-01-05:
@DonLakeFlyer was it you who was working on this?

DonLakeFlyer, 2016-01-05:
The travis part is done. You just need to update the param meta data parser to catch a validation error and the build will fail. What exactly do you want to verify. I'm not quite sure.

magicrub, 2016-01-05:
Not sure, this issue was generated after a dev call back in May and it hasb't been touched. I was looking through them and realized this is what you're been doing. Although not exactly since this is sort of param specific (of something still not implemented. I'll go ahead and close it. Thanks for working on it!

closed 2016-01-05 by magicrub

DonLakeFlyer, 2016-01-05:
Here us what I had planned:
- Validate correct @Values/@Bitmask string format. Commas and colons in right places for example.
- Validate that anywhere there is supposed to be a numeric value, that the string is really a number.
- Validate that the units designation comes from a known set. The reasons for this is that QGroundControl automatically translates some units to more user friendly units. In order to do that we need high quality on the units meta data.
- Validate that @DisplayName is not too long

DonLakeFlyer, 2016-01-05:
You can leave it open, assign it to me and I can use it to track my work.